gnea/grbl #999: HandHeld MPG Encoder ?

rodrigoevaldo, 2021-02-04:
I need some suggest for add the manual MPG encoder, for JOG movement?

My application I have two arduino Mega. One arduino running the GRBL 1.1 and another for other simple operations, like safety interlock for example.
The second arduino, I added the manual encoder for counting the pulses for manual movement. How I can send the commands to first one?

Today, I implemented the Computer read the second arduino the total pulses. The computer by USB/Serial send commands, for example "G0 Xnn" (nn is the pulses from encoder). How the computer have a communication delay the result are not good.

SailWithChips, 2021-02-04:
Hi rodrigoevaldo, although this is not an issue an should be closed or posted elsewhere have a look to this link... there are lot of diferent diy sollutions some of this could help your idea. Look at this: https://wiki.shapeoko.com/index.php/LCD_on_GRBL

terjeio, 2021-02-05:
Here is [another project](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack), I added a second serial input to the Mega to make it work.