gnea/grbl #1001: Spindle or z-limits, can't get both

dmb0058, 2021-02-06:
Hi,

I suspect this has been done to death but I can't get my head around it at all.

I'm using GRBL 1.1h on an Uno, with a CNC shield v3.51. I have connected six limit switches, two on each axis, to the Z+/-, Y+/-, X+/- pins on the shield. These appear to be duplicated, e.g. Z+ is connected to Z- etc. I've also connected SpnEn to the control pin on a relay, which switches power to a single-speed spindle.

If I comment out the "#define VARIABLE_SPINDLE" line in config.h and upload GRBL, the z-limit switches work but the spindle enable does not ($21=1), i.e. closing either z-limit switch triggers an alarm in Universal G-Code Sender, but "M3 S1000" and "M5" do nothing.

If I uncomment the line, then "M3 S1000" enables the spindle, "M5" turns it off but closing the z-limit switches has no effect (the X and Y limit switches work fine).

So I'm stuck with spindle control or z-limit switches but I can't see how to get both ... 

Any help appreciated,

David

PicEngraver, 2021-02-07:
Use 1.1h as it was released without making any source code changes.  The board is an old design for early .9 grbl use, so the pin markings are wrong in the Z+ limit (End Stop) and spindle markings for 1.1 grbl.

The Z+ End Stop pins are for the PWM (spindle) output signal, and the Spindle Enable pins are for the Z+ End Stop input.  This is what works on my board with no source code changes.  I do not use the Z- limit pins, so assume they are OK as marked.

dmb0058, 2021-02-07:
Hi John,

So I think what you're recommending is to leave "#define VARIABLE_SPINDLE"
uncommented as it comes, then connect my spindle control line to the pin
marked Z+ and my z-endstops to the pin marked SpnEn on the 3.51 board
instead of the other way around ?

Sorry if I'm being stupid :) It's just that with that line uncommented my
z-endstops work 100% connected to the pin marked Z+, so it seems strange
that I could just plug the spindle control into the same pin, which would
also work as a spindle control without changing anything else. I'll give it
a go anyway ... I'm probably misunderstanding something.

Thanks,

David

PicEngraver, 2021-02-07:
Best you do some Google searching to see if your shield version was changed to set it up correctly for grbl 1.1.  My searching showed it set up for grbl 0.9, which I believed used the old pin connections on the Arduino for grbl 0.8 and earliest grbl 0.9???  Using a continuity meter to verify pin connections between the shield and the Auduino is probably best to do.  I had to wire my ver. 3.0 board as I said for grbl 1.1, so assume you need to do it, too, from your description.

dmb0058, 2021-02-07:
Hi John,

Yes I'm sure you're right that it's 0.9 compatible, so should match your
experience.

I've torn Google inside out for the last couple of weeks but I can't figure
out how this combination of shield and software works from the sparse
documentation available. I can see loads of similar queries but none that
actually report how they got it working ...

I'll stick it on a meter and see if I can make sense of it all, and then
see if I can write up a simple page that explains how to get a 3.51 shield
working with GRBL 1.1h for any future unfortunate souls with the same
question :)

It's fantastic that this software and hardware is so freely and cheaply
available even with the effort required to get it going.

Thanks again for your help,

David