gnea/grbl #1002: Homing - X Axis not zeroing

TrutNoone, 2021-02-15:
Hi, I'm finishing up on my build and have run into another small problem. Whenever i home my machine (industry standard x+ y+ Z+), Y and Z show -5 machine coordinates as expected but X shows 190 instead of -5. It appears to not be zeroing x axis properly. My x axis max travel is 195mm and it reports 190 after the homing offset.

I am pretty new so excuse me if I'm wrong, but shouldn't machine coordinates be -5 -5 -5 not +190 -5 -5.

Thanks for any assistance

PicEngraver, 2021-02-17:
Generally, you can safely just ignore the machine coordinate settings.  You will normally do your projects using the work coordinate settings, which is where you set your bit to X0/Y0/Z0 on your material when beginning a project.

TrutNoone, 2021-02-17:
Solved, not sure what was causing it to do that, but a re-flash of firmware and clearing of eeprom solved it. Thanks

closed 2021-02-17 by TrutNoone