gnea/grbl #1004: Arduino Nano Every problem

nikryden, 2021-02-16:
I have no luck to upload the hex to Arduino Nano Every (ATMEGA4809) from Windows 10
I have used avrdude but it won´t work for me. when I upload any examples from Arduino UI it works.
Is there any one the have succeed to upload to Nano Every that can give me any pointers it feels like I tested everything on Google.

**This is the command line I use**
```
 "C:\Users\ryden\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" -C "C:\Users\ryden\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -p atmega4809 -c jtag2updi -P COM7 -b 115200 -e -D -U flash:w:"C:\temp/grbl.hex":i -U fuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m {upload.extra_files}
```
**Result:**
```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\ryden\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM7
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): timeout/error communicating with programmer (status -1)

avrdude done.  Thank you.

```

109JB, 2021-02-20:
In the arduino UI, select "Atmega 328 (Old Bootloader)"  under Tools/Processor

nikryden, 2021-02-20:
Thank you 109JB for your fast help, 
but in my case that did not worked as the _Arduino Nano Every_ not use 'Atmega 328' but the 'ATMEGA4809' .

But I manage to find the solution and put it in a bat file.
The solution was to forcing reset using 1200bps open/close on port COM(n) in my case COM7. 

This is the Bash that I created(this is for Windows)
```
@echo off
c:\windows\system32\mode.com com7: baud=1200 dtr=on
c:\windows\system32\mode.com com7: dtr=off
"C:\Users\ryden\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" -C "C:\Users\ryden\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -p atmega4809 -c jtag2updi -P COM7 -b 115200 -e -D -U flash:w:"C:\temp/grbl.hex":i -U fuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m {upload.extra_files}
pause
```

SimonJohnCastle, 2021-02-22:
If someone wants to pick this up, what did you do already to compile for the correct processor.

Priedon, 2022-03-21:
Hi @nikryden ,

Did grbl work on the Nano Every correctly? Have you encoutered any issues with it?
Is it possible to use the Arduino IDE to upload grbl to the Nano Every?

nikryden, 2022-03-22:
Hi @Priedon, 

It works fine but I did not upload it with Arduino IDE I used the bat file. 
Sorry I have not tried to upload with Arduino IDE after I made the solution with the batch file. 

The solution for me was to forcing reset using 1200bps open/close on port COM(n) in my case COM7. 
If you can do that in the Arduino IDE sorry I don't know. 

**Bash file:** uploadHex.bat
```
@echo off
c:\windows\system32\mode.com com7: baud=1200 dtr=on
c:\windows\system32\mode.com com7: dtr=off
"C:\Users\ryden\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" -C "C:\Users\ryden\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -p atmega4809 -c jtag2updi -P COM7 -b 115200 -e -D -U flash:w:"C:\temp/grbl.hex":i -U fuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m {upload.extra_files}
pause
```
The bash will only work on windows.
The ``` flash:w:"C:\temp/grbl.hex" ``` is the path to the hex file. The hex file do you find at https://github.com/gnea/grbl/releases

To find path to avrdude I did this :
_Open Arduino IDE_

- File > Preferences > Show verbose output during: > compilation (uncheck) > upload (check) > OK
- Sketch > Upload  
- After the upload finishes/fails (doesn't matter which), scroll up the black console window at the bottom of the Arduino IDE window until you find the avrdude command the Arduino IDE generated for the upload. That will show the location and also provides a reference command you can adapt to your needs.

litherw, 2022-04-01:
Hi @nikryden ,

I've adapted your script to upload GRBL to an Arduino Nano Every board and it looks like the script ran okay, but when I launch UGS it doesn't recognize GRBL is connected.  I downloaded the .hex file from the link in your post above, but is that the correct file, or are there other steps needed to prepare a .hex file for the ATMEGA4809? Have you any troubleshooting tips? 

Thank you

nikryden, 2022-04-01:
@[litherw](https://github.com/litherw) I don't use UGS so I can't help u there. I am using CNCJs and http://chilipeppr.com/ and it works with them

I was some time ago so I don't remember if I compiled it special for ATMEGA4809, but here is what I used http://gofile.me/4MzNh/CR3u4j2oj . the link is only available to 15 of April 2022.
Files:
grbl2.hex, upload2.bat
you must edit upload2.bat

litherw, 2022-04-01:
@nikryden Thank you for the quick reply.  I uploaded your grbl2.hex with the avrdude script and am still getting the same result - bCNC, UGS, and chilipeppr all connect and recognize the Nano Every board serial port, but cannot connect to GRBL (no response / shows offline).   Maybe my serial port baud rate is wrong...

I will keep troubleshooting, and continue to use my original Nano GRBL for now.  Thank you again.

nikryden, 2022-04-01:
@[litherw](https://github.com/litherw) I found a grpl4.hex file on my hard disk that is slightly different and newer and smaller that you can test. I don't remember if that was the file I finally used, I tough it that was the grbl2.hex I used, but then I found the newer grpl4.hex file in a child folder. Maybe worth a try.
http://gofile.me/4MzNh/p0IsOvObh

markmcgookin, 2022-09-09:
Came across this myself using Arduino IDE on linux. @109JB s answer was almost right for me. But for me it was to do with emulation. The fix for me was "Tools > Registers Emulation > None (ATMEGA4809)"