gnea/grbl #1007: F word meaning,

frett27, 2021-02-19:
Hi there, we are willing to known the meaning of the number value associated to a gcode F word using GRBL. We've not found any detailed explaination for the meaning of this values. 
 
Does this value is associated to speed limits, $110 / $111 how does is play with it ?

any response is welcome, and if we can contribute to help on the documentation, 
thank's for your attention on this important point for us.

Patrice

idiydownunder, 2021-02-20:
I think you'll find that the gcode F sets the feed rate for that move or until it's changed again. The settings in $110/$111 will allow any speed up to the set feed rate, but will not let the feed rate excide the limit set in $110

frett27, 2021-02-20:
hi, @idiydownunder , thank's for your precision and response, it helped. So do i have to understand the feed is expressed in mm/min, as the $ settings ?

neilferreri, 2021-02-20:
Assuming G94, the feed will be in units/min.
The unit depends on the G21 (mm) or G20 (in).
Your $110/$111/$112 are always in mm/min.

frett27, 2021-03-01:
@neilferreri yes we are in ISO metrics, 
@neilferreri  @idiydownunder  thank's for your help, i could adjusted the behaviours of GRBL controlling software and settings, accordingly to your inputs.

closed 2021-03-01 by frett27