gnea/grbl #1014: Override a movement command while it is still running?

Elmi77, 2021-03-02:
Hi,

assumed I already have started a movement e.g. by sending a G0 command. Now while this movement is still in progress and axes are still running, my application decides to not to go to the target position that was specified by this first G0 command but to a different target-coordinate.

Is this possible? Can I send a second motion-command which replaces the target of the first one (while it is still moving) so that the axes no longer travel to the first target coordinate but immediately try to reach the second one?

If yes: how can this be done exactly?

Thanks :-)

Sortland, 2021-04-13:
Seems to be a really useful function - so I would like to second this as a feature request!

zeevy, 2021-04-16:
May be not possible, currently you have to stop the machine (using soft reset) then send the new target

Sortland, 2021-04-19:
Yes, that's why it is a feature request for a new feature!