gnea/grbl #1015: Cutting proble along X-axis Ortur Laser aster 2

mtvjulya, 2021-03-02:
Hi. I have recently bought Ortur Laser master 2 and having issue with lines not matching on X and Y axis.
I have tried readjusting focus multiple times, cleaning the lens, tilting laser sideways but no luck. Y axis cuts clean and deep while X axis is making a broad line with much less penetration… Can somebody help me to solve this problem?
[image]
[image]

Elmi77, 2021-03-03:
I think this would be more a question for the support of the vendor of this device.

Klaus3777, 2021-03-08:
Hello, please rotate your wood 90 degrees and test again, it could be that the layers of the wood can only be cut quickly in one direction