gnea/grbl #1027: Getting Errors with laser - hitting limits somwhere?

bz6djs, 2021-03-20:
Hi All, I use a Neje Master 2 laser with a 30W head. I run GRBL 1.1f on the 8bit board and the laser has a bed that is 170mm x 170mm in size.  I believe I am running into some limit but I am unable to figure out where or how to modify that limit or bring the image I am working with into the limit. I am hoping someone here can explain it to me as I do wish to learn more so if I encounter this again I will know what I am looking at.  I turned on verbose logging in the console and I see this (these are truncated so if you need more pls let me know)
When I turn on the laser I see this:
Waiting for connection...
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
error:9
G-code locked out during alarm or jog state.
[MSG:Caution: Unlocked]
ok
[VER:1.1f.20170801:]
[OPT:V,15,128]
Target buffer size found
ok

after it begins to process my job I see this in my list 
G1X-0.17S67.8
`fx��`F`

G1X-0.08S93.9
`fx��`fx

G1X-0.34S0
`fx��`fx

G1X0.08S76.5
`fx��`fx

G1X-0.08S83
`fx��`fx

G1X-0.08S72.9
`fx��`fx

And now the head stops moving.   I will copy the entire file here
[new 2.txt](https://github.com/gnea/grbl/files/6175715/new.2.txt)

I would really like to understand what limit I am hitting and what I may need to change to be able to get past this so I can create images etc.  Any assistance you can offer would be appreciated

PicEngraver, 2021-03-23:
It seems you have lines in your gcode that grbl can not process.
"Layer C00"
"fx���fx"

I do not know if your sender sends these lines to grbl or skips over them.  There may be other issues with your gcode file also.

bz6djs, 2021-03-23:
Any of the codes being sent would be coming from Lightburn, which is the software I am using.  I started a thread with them and was told the issue may be earthing ground .... and perhaps there was static running through my board so I grounded the laser in several locations to be sure of its ground.  it would appear that layer C00 is the image layer as per this 
[image]

Would you know of any way for me to remove these errors?  I am running GRBL V1.1F  ...

Thank You for responding,
Ron

PicEngraver, 2021-03-23:
Sorry, but I do not use, have never used, Lightburn.  They are a competitor of sorts. www.picengrave.com

I do assume that whatever Lightburn adds to your gcode would be correctly managed by it for grbl usage. 

John