gnea/grbl #1028: Weird Values set as machine position after homing...

simrim1, 2021-04-07:
Hi,
I have spent a lot of time searching on google for starnge behavior shown by GRBL. I got Arduino UNO (Clone) and have flashed 1.1f through Arduino IDE without any change in the configuration file(s). The settings are as follows.

$0 = 5    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 3    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 19    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
**$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 1    (Homing direction invert, mask)**
$24 = 200.000    (Homing locate feed rate, mm/min)
$25 = 1500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 100.000    (X-axis travel resolution, step/mm)
$101 = 100.000    (Y-axis travel resolution, step/mm)
$102 = 160.000    (Z-axis travel resolution, step/mm)
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 600.000    (Z-axis maximum rate, mm/min)
$120 = 200.000    (X-axis acceleration, mm/sec^2)
$121 = 200.000    (Y-axis acceleration, mm/sec^2)
$122 = 100.000    (Z-axis acceleration, mm/sec^2)
$130 = 404.000    (X-axis maximum travel, millimeters)
$131 = 485.000    (Y-axis maximum travel, millimeters)
$132 = 52.000    (Z-axis maximum travel, millimeters)

As you can see I have inverted X a-axis homing direction only. It does homing process correctly i.e. stops where I want it after finding homing switches. The issue is setting the machine position. It sets Y and Z to -2mm (because of 2mm homing pull-off distance) but it sets X axis to somethimes at -402 or -204. I think it should set it as 2mm (positive value) because X axis homing direction is inverted. I don't know where it is picking up -402 or -204.

I will be extremely grateful for any help in pointing out what I am doing wrong. I am using UGS classic 2.0.7 on windows 10.

Thanks
[image]

simrim1, 2021-04-09:
I think I have figured out why X machine position (with inverted home direction flag in $23) is not setting at 2mm (or whatever is pull-off value) after hitting the physical limit switch. It is playing with the value of $130 (x-axis maximum travel). The code starting line 379 in limits.c is as follows

if ( bit_istrue(settings.homing_dir_mask,bit(idx)) ) {
     set_axis_position = lround((settings.max_travel[idx]+settings.homing_pulloff)*settings.steps_per_mm[idx]);
} else
{
     set_axis_position = lround(-settings.homing_pulloff*settings.steps_per_mm[idx]);
}

As you can see if the homing direction is not inverted then it doesn't bother with settings.maxtravel[n] which is $130-$132 depending on the axis in question.

Strangely enough, if #HOMING_FORCE_SET_ORIGIN is enabled then Machine Position is simply set to zero for all axis regardless of the homing direction bit. Code line 376 in the same limits.c file.

@chamnit I will be grateful to you if you can shed some light on the necessity of using maximum travel in the calculation of Machine Position only when the homing direction is inverted and #HOMING_FORCE_SET_ORIGIN is disabled.

I have compiled grbl remving settings.max_travel[idx] from the calculation and now it sets Machine Position at 2mm but I don't know if it will break something else while running G-Code.

Any guidence/information/advice will be highly appreciated.

chamnit, 2021-04-10:
Please see the homing setup documents. The homing direction setting set which side of an axis the limit switch is on. It also assumes a Cartesian coordinate frame that follows the right hand rule.

RonaldDegenkamp, 2021-06-29:
The issue stated is the right way for GRBL to work.
For X axis homing it moves all the way to the left. Sets the position to the Max x value  and pulls of 2 mm.
max x asis =-404 as stated in $130 = 404.000 (X-axis maximum travel, millimeters) 
2 mm retract means X=-402.