gnea/grbl #1030: Wiring Limit Switches - location of capacitor in optocoupler circuit

tmzqvd, 2021-04-12:
I know it's a trivial question. But I got confused following instructions on [Wiring Limit Switches](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches).  Can someone please clarify the location of capacitor. NC switch configuration. Two separate power supplies. 

First option:
[image]

Second option:
[image]

Or none of them?

Thanks.

lalo-uy, 2021-04-13:
The cap must go on the transistor side of the opto, to slow down
transitions.

tmzqvd, 2021-04-13:
Do you mean like this?

[image]

This was my first idea. But then I read this post [limit switch trigger report](https://github.com/gnea/grbl/issues/852) (user biletnikov stated "In your case, I would suggest to add 0.1 uF capacitor in parallel to the contacts of switch.") and the schema he made [Limit Switch board for CNC machines](https://hackermagnet.com/limit-switch-board-for-cnc-machines/). 
I don't like the way he's wiring switches therefore I want to make my version.

lalo-uy, 2021-04-15:
Yes, that is the way. When opto goes on C1 discharg. As it goes off C1
charge from 10k R1 and takes more than 1mS .

tmzqvd, 2021-04-15:
Thanks. What about putting capacitor on both sides of opto? Does it make any sense? If yes how to put it on left side? First or second option? 
Let me explain why I think it's good to have it on switch side (left) of opto. Switch is on cnc machine and wires goes with other wires  and also there is a spindle motor. Opto will be at the end of switch wires, next to arduino. Therefore I think problems are generated on that side. Maybe it's best to prevent false signals even comming to opto. 
I will not put this circuit next to switch, but next to arduino. They will be on the same pcb.. Relative long wires will come from switch to this circuit.

tmzqvd, 2021-04-16:
I added another option - 4. I think these are all. At present moment I'm leaning towards 3 and 4 combined. What do you think?

[image]

It's obvious I don't have enough theoretical knowledge.

RonaldDegenkamp, 2021-06-04:
Only option 4 of the above makes any sense.
The only reason for the capacitor is shorting the bounce of the switch.
The only place it can do that is directly parallel to the switch,

RonaldDegenkamp, 2021-06-04:
i read you are using this circuit on an arduino, Use the pinMode ( pinNumber , INPUT_PULLUP ); and  lose the 10K pullup resistor R1 in the schematic of option 4.

tmzqvd, 2021-06-08:
Thanks @RonaldDegenkamp. Is INPUT_PULLUP reliable? Is it working on nano? I have never used it. I'm always using classical INPUT with added resistor. 
I will use grbl and I'm not sure it's using pinMode ( pinNumber , INPUT_PULLUP ). Or maybe I can set it to work this way. Have to check.

RonaldDegenkamp, 2021-06-29:
INPUT_PULLUP is the most reliable way and works on a nano.
GRBL uses INPUT_PULLUP on all input pins exept 13 where the onboard led is connected (on UNO)

graham2550, 2022-10-16:
The 5v supply and the associated  GND on the switch side of the optocoupler must be a separate supply otherwise there is NO isolation. This has to be a completely separate supply.