gnea/grbl #1036: grbl laser mode, how many motion commands per second?

svenhb, 2021-04-29:
How many motion commands per second can be processed on an Arduino uno?

#### Extreme example: 
feedrate = 6000 mm/min = 100 mm/sec
raster 0.1 mm (needed GCode in G91 e.g. G1X0.1Y0.1S10)
Would need 1000 Gcode lines/sec = ca. 14000 Bytes/sec.

I found a limit at ca. 100 Gcode lines/sec = ca. 1000 Bytes/sec (feedrate 4000 mm/sec = 67 mm/sec at 0.4 mm raster with ca. 8 Bytes)
Image paint starts at 4:30 https://youtu.be/7Nvf4UwK_yk?t=269
Unsmooth motion because of too high feedrate starts at 5:08 https://youtu.be/7Nvf4UwK_yk?t=308

Or is the USB port the limitation?

Norwegian-Gadgetman, 2021-04-29:
The serial port allows for a transfer rate of up to 112500bps.

But this comes down to 1start bit, 8 data bits, and 1 - 2 stop bits per transferred Byte.
At 10bits, you can transfer 11250 Bytes / Second... in theory...
but that's theory...
This assumes a lot of timing matches perfectly, and that the Arduino can make use of everying as it gets it.
Lets just say that on a good day, the max is around 10KB.

What you need to do is look for ways to decrease the amount of data that needs to be transferred.
If you run absolute coordinates, look for G0/G1 commands where one parameter doesn't change from one command to the next.

If the X or Y coordinate doesn't change, there's no reason to include it in the second command.
Decreasing the accuracy may also help. There's usually no need for more than one digit after the '.'

DOS/Windows based programs usually add both [Cr] and [Lf] to the end of each line. Getting rid of the [Lf] symbol will also help slightly. (This may net you anything from 2 - 5% improvement )

Trygve

svenhb, 2021-04-29:
>What you need to do is look for ways to decrease the amount of data that needs to be transferred. 
 
Yes I did this already: using G91, only send new coordinate if S needs a change (ok I send same S on start and end of a route to avoid ramps, when using Z instead of S).

terjeio, 2021-04-29:
> The serial port allows for a transfer rate of up to 112500bps.

The rate is 115200 so a little bit more ;-)

Two of the main limiting factors (in addition to the baud rate) is sender transfer method (single line/wait for ok or keeping the input buffer full) and planner buffer size (max 256 entries in Grbl but limited to available RAM, default is 36?). See [this grblHAL issue](https://github.com/terjeio/grblHAL/issues/142)  and [this blog post](https://embeddedtronicsblog.wordpress.com/2020/07/20/grblhal/) for more info.

svenhb, 2021-04-29:
I didn't mentioned: I used the Character-counting Streaming Protocol, and the buffer size is 127.
And I'm using "classic" grbl on Atmega 328 of Arduino Uno...

terjeio, 2021-04-29:
> the buffer size is 127.

This is the serial input buffer size.

The planner buffer size affects the max. throughput quite a bit - at least at high feed rates.
And it is has even less entries than I remembered. You need a processor with more RAM to speed it up?

https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/planner.h#L26-L33

svenhb, 2021-04-29:
>You need a processor with more RAM to speed it up? 
 
No :-) I'm more interested to calculate or estimate the max. amount of commands per second...

langwadt, 2021-04-29:
try running at, say, half or double the baudrate to determine it is if the communication that is the limit or the AVR simply doesn't have the horse power to process the moves any faster

terjeio, 2021-04-29:
> No :-) I'm more interested to calculate or estimate the max. amount of commands per second...

To do that you have to sort out where the bottleneck is. IMO it is likely to be the planner buffer. Perhaps you can get at an estimate by sending a stream of motion commands that just about keeps the planner buffer filled? Or check in your sender when ok responses arrives after a delay? (it will be when the planner buffer is saturated)

From what we found streaming gcode in character counting mode did not factor into this until a rather large planner buffer was used. Another factor that may play into this is the processing speed of the 8-bit CPU, it is rather slow (compared to M4F ARM) and the planner eats a lot of cycles...

If you have access to a controller with more RAM (the 2560 Mega?) you may play with the size of both the input buffer and the planner buffer sizes to get a feeling of how those affects the throughput?

> the AVR simply doesn't have the horse power to process the moves any faster

There may be horse power but too little RAM to work with?

svenhb, 2021-04-29:
It seems: I need to figure it out by myself...
1st I will try to log the planner buffer (Bf:15,128. The first value is the number of available blocks in the planner buffer and the second is number of available bytes in the serial RX buffer.)

manotroll, 2021-06-12:
can you increase the serial communication speed?
or write the data without booting to free up space