gnea/grbl #1038: 1.1 H dual axis homing: Z strange behavior

BernardG, 2021-04-30:
I did setup for 2 motors homing on my Z axis, and it "seems" to work OK. But now, if I start a homing process with my Z far from the top (say 80mm below 0 in machine Pos), it goes up a little bit (maybe 10 mm), and I have to start homing something like 8 times before it finally touches the switch, and continue on homing Y and X, which then works as it should.

Any idea about what could cause this?
Thanks for your help

J-Dunn, 2021-05-25:
I just read in the release notes that dual homing was only supposed to work on X and Y, so what you are doing maybe unsupported/untested. Have you tried homing X and Y first? Just an idea.

BernardG, 2021-05-26:
> I just read in the release notes that dual homing was only supposed to work on X and Y, so what you are doing maybe unsupported/untested. Have you tried homing X and Y first? Just an idea.

At first, I was surprised by your answer, then realized I made a typo. It's on the X axis that I have 2 motors, not Z. Sorry! I corrected that typo, now.

J-Dunn, 2021-05-27:
OK that makes more sense. 
As a diagnostic, how about only doing X,Y or only doing Z homing.  See config.h 
```
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.

```
( I've defined constants to make it more readable, IIRC std grbl is 0,1 and 2. )

Does Z home properly if that is the only one ?

BernardG, 2021-09-10:
> OK that makes more sense.
> As a diagnostic, how about only doing X,Y or only doing Z homing. See config.h
>
> ```
> #define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear [...]

I come back to this, as this is really annoying. I have exactly what you copied in my config.h.
It looks like, when homing Z, it makes an error "homing failed", like if the values for triggering an error in case of the homing was too far apart for the 2 X motors, was taken into account on Z.

#define DUAL_AXIS_HOMING_FAIL_DISTANCE_MAX  25.0  // Float (mm) was 25

Obviously, that cannot be right! Could it be something in my wiring ? Does anyone else uses this feature properly ?

J-Dunn, 2021-09-10:
I can't test because I use a modified 0.9j build. 

What happens if you just have HOMING_CYCLE_0 defined and thus just test Z alone. The two cycles should be totally independent, so make sure Z home works first. Its seems that it where your problem is.

BernardG, 2021-09-11:
Maybe I am not clear. They DO work independently. BUT, Z homing stops every 25 mm or so, saying homing failed, until I start homing enough times so that it hit the limit switch in the last +- 25 mm up.

To say it differently, if I start homing with Z axis less than 25 mm away from the switch, it works as it should. If not, it goes up 25 mm, trigger an error, etc...

BernardG, 2021-09-11:
Well, OK, I am not saying I understand, but I am not sure this is right. If I manually trigger Z, I get

```
> ?
<Idle|MPos:-469.001,-259.000,-21.000|Bf:15,127|FS:0,0|Pn:PXZ>
```

and I get the same if I manually trigger both X1 and X2, which then seems normal.
Am I confused, or did I made a mistake wiring my switches?

J-Dunn, 2021-09-11:
If you trigger an end-stop it should put the machine ( grbl ) in an alarm state which requires a reset,  plus an alarm message saying you hit an end stop.  ( Well , since 1.0 it will give you a dumb error code which you have to decrypt somewhere else.). 

This is turning into a guessing game. You probably need to state clearly your hardware and software configurations at this stage. 
What limit/homing switches do you have and post your config.h

BernardG, 2021-09-11:
> If you trigger an end-stop it should put the machine ( grbl ) in an alarm state which requires a reset, plus an alarm message saying you hit an end stop. ( Well , since 1.0 it will give you a dumb [...]

I am sorry, but... you're wrong, and seem to not know GRBL well enough to help. What you say would be right if triggering the end-stop was done WHILE executing G-Code, but it's not true if you do this while the machine is idle.

J-Dunn, 2021-09-11:
Oh well, guess you're on your own then. Good luck.

langwadt, 2021-09-12:
@BernardG if you have hard limits enabled you get an alarm if an endstop is triggered even in idle state

BernardG, 2021-09-12:
I don't have hard limits enabled, as it is not a good choice, IMHO, BECAUSE it triggers an alarm.... :-)

GrahamJB, 2021-09-12:
What is your $132 setting?

BernardG, 2021-09-12:
> What is your $132 setting?

`$132=195.000 (Z-axis maximum travel, millimeters)`

GrahamJB, 2021-09-12:
So that's not the cause of the problem then!

BernardG, 2021-09-12:
> So that's not the cause of the problem then!

haha ! No, it does not seem to be. If it was that obvious, I don't think I would be asking for help here..... ;-)

GrahamJB, 2021-09-12:
I apologise for suggesting a possibility, and that I seem to have offended you!

BernardG, 2021-09-12:
> I apologise for suggesting a possibility, and that I seem to have offended you!

No, no, it did not offend me, just made me smile. Sometimes, written communication is not easy, and English is not my first language. It's just that I believe I have looked at all possibilities, without finding what I could have done wrong.