gnea/grbl #1043: Z axis limit switch produces multi-alarm error with cnc shield

bluejellybean, 2021-05-21:
Hello, I am building a custom cnc and it is in the final stages of wiring and configuring. Unfortunately, I am running into an error that I have been unable to diagnose for a few nights now despite heavy trial and error, and googling. I am using an arduino uno, a cnc shield, and grbl 1.1. I have the limit switches wired for x+, x-, y+, and y-, all are working as intended.. when I depress each, only 1 alarm for that axis is triggered. The z-axis + and - on the other hand, are incorrectly displaying alarms for x and z, and x and y, respectively,  while depressed. I have tried configuring the limit switch to use pin 12 with variable enabled, as well as pin 11 with variable disabled as per the v1.1 requirement.

I'll be trying a new shield tomorrow, and if that doesn't work, I'll be digging into the grbl codebase more thoroughly, but I wanted to check here first to see if anyone has experienced anything similar. Cheers and thank you for the hard work on this project, I'm looking forward to getting everything configured properly.

bluejellybean, 2021-05-24:
Some progress, I think this may be due to an issue with my uploader not correctly updating the arduino. I did a quick test and part of my issue has been resolved (z-limit correctly triggers without x limit also triggering). I'll have to rewire things and test the complete setup before I can completely confirm, should be sometime this week.

frankrieder, 2021-05-24:
I also had the same issue, I just chalked it up to me doing something wrong, maybe its time to redo my arduino upload.