gnea/grbl #1046: laser machine moving in two axis

marcootaviomello, 2021-06-02:
I'm modifying a laser machine, which didn't use grbl firmware, used proprietary software along with proprietary company firmware. I recorded the GRBL on it, but when testing with the LASER GRBL when I click on the diagonal movement arrows, that is, it has command in both axes, it moves correctly, but when I click on the conventional up, down, left, right buttons, the same moves diagonally. ME seems to be something from mechanical construction, but can I solve it via GRBL configuration, to send pulse to both engines?

langwadt, 2021-06-02:
how does the machine look?

marcootaviomello, 2021-06-03:
Analyzing the mechanics, this machine has only 2 motors, to make movements in a straight line I must send a command in both axes, (which is when we click on the arrow to move diagonally on the LASER GRBL), if I send a command only in one of the axes, the machine moves diagonally to the command axis. Any configuration ideas?
[image]
[image]
[image]
[image]

langwadt, 2021-06-03:
rebuild with COREXY defined, see config.h line 182-189

marcootaviomello, 2021-06-07:
Thanks a lot.