gnea/grbl #1049: soft limits and probing

justme-1968, 2021-06-26:
i'm just exploring soft limits and have noticed that after enabling them probing with G38.2 stopped working. it looks like the z distance is checked against exeding the z soft limit before the motion is started. 

is this the expected behavior? i think this makes it impossible to have a generic probe command configured that uses most of the z travel. now i have to make sure the spindle is as far away from the bottom z soft limit as possible or i have to use a small z travel in the probe command and maybe use it multiple times if the spindle is too far away from the probe.

would't it be better to check during the motion so that the probing can run?

RonaldDegenkamp, 2021-06-29:
Hey justme, It is the normal behaviour of GRBL to reject commands before they can move the machine outside soft limits.
I had the same problem now i just set the Z axis 5 mm above the probe hight and have no problem at all.

justme-1968, 2021-06-30:
ok. thanks. being new to this it was quite unexpected. in the mean time i had the same experience with jog motion if the increment is too big.

closed 2021-06-30 by justme-1968