gnea/grbl #1050: Lead CNC Y- Axis homing issues with UGS

JonnayLin, 2021-07-03:
I'm having troubles with the software/electronics on my Bulkman 1515 Lead CNC. I'm using DM542t drivers and an arduino nano with grbl 1.1h installed. For software, I'm using Universal G-Code Sender. 

When I'm trying to home the machine, the Z-Axis and the X-Axis work perfectly fine, but when it goes to the Y-Axis, it does this weird stepping in the wrong direction then sends an error code. My endstop switches are wired in normally open and all the axis move well and in the right direction if I'm jogging them manually after a reset of the machine. Thanks in advance!

A video showing the issues is listed here: https://youtu.be/Nq9MnFdqH1M

The firmware settings for grbl that I'm using are listed at the bottom of the page in the link: https://bulkman3d.com/knowledge-base/high-current-grbl-cnc-controller/

PicEngraver, 2021-07-03:
I hope this will help you diagnose switches.  Should work with UGS.
Limit Switch Testing
John Champlain
2/17/2021
[image]

JonnayLin, 2021-07-04:
Do I have to get the paid version to access this function? I have the test version right now and I home the machine, it shows that it's ready, but I am not able to select the $ option or jog the machine.

PicEngraver, 2021-07-04:
If you are trying the demo version of my program, PicSender, it should allow you to connect to grbl and use the "?" button as described for testing the switches.  Does it not work for you?  The testing procedure should also work in UGS for you.  Please give more details of how you are testing.  It may be easier to assist you if you email me direct.

JonnayLin, 2021-07-04:
I messaged on the other thread, but I decided to go with the openbuilds blackbox platform after struggling with this problem for the entire day yesterday....hopefully it'll work!