gnea/grbl #1051: Lead CNC Homing Issue

JonnayLin, 2021-07-04:
Hello, I'm having some issues with my homing with Universal G code Sender. I built a Bulkman Lead CNC 1515 with DM542t drivers and the arduino nano running 1.1h. After I home the machine, it comes back with the following: 

An unexpected error was detected: (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.
[MSG:Reset to continue]

If I don't home the machine, and I set the hard limit $21=0, I'm able to jog the machine manually in all directions. My switches are wired in normally open, but I have also tried with normally closed and I still have the same issue. Thanks in advance!

PicEngraver, 2021-07-04:
The most likely cause is EMF spikes on the switch wiring.  Very, very, very common, and difficult to remedy (requires voltage suppression circuitry be added to the switch wiring, and this is not always a 100% fix).  I recommend the use of soft limits ($20=1).

JonnayLin, 2021-07-04:
After trying to shield the wires, it was still happening. I just finally decided to try openbuild's blackbox platform. Hoping that will be the easiest solution that will just work! Thanks for the help.

PicEngraver, 2021-07-04:
Good luck.