gnea/grbl #1052: Error Compiling "warning: "max" redefined"

CobaltEcho, 2021-07-07:
I've tried this several times, watched videos, re-read the instructions, followed step by step.  And every time I'm getting this error:

On an Arduino Nano and using Arduino IDE 1.8.15

`In file included from C:\Users\Echo\Documents\Arduino\libraries\grbl/grbl.h:43:0,
                 from C:\Users\Echo\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:
C:\Users\Echo\Documents\Arduino\libraries\grbl/nuts_bolts.h:51:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 
In file included from sketch\grblUpload.ino.cpp:1:0:
C:\Users\Echo\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 
In file included from C:\Users\Echo\Documents\Arduino\libraries\grbl/grbl.h:43:0,
                 from C:\Users\Echo\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:
C:\Users\Echo\Documents\Arduino\libraries\grbl/nuts_bolts.h:52:0: warning: "min" redefined
 #define min(a,b) (((a) < (b)) ? (a) : (b))
 
In file included from sketch\grblUpload.ino.cpp:1:0:
C:\Users\Echo\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition
 #define min(a,b) ((a)<(b)?(a):(b))
 
In file included from C:\Users\Echo\Documents\Arduino\libraries\grbl/grbl.h:43:0,
                 from C:\Users\Echo\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:
C:\Users\Echo\Documents\Arduino\libraries\grbl/nuts_bolts.h:55:0: warning: "bit" redefined
 #define bit(n) (1 << n)
 
In file included from sketch\grblUpload.ino.cpp:1:0:
C:\Users\Echo\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/Arduino.h:124:0: note: this is the location of the previous definition
 #define bit(b) (1UL << (b))
 
Sketch uses 29594 bytes (96%) of program storage space. Maximum is 30720 bytes.
Global variables use 1484 bytes (72%) of dynamic memory, leaving 564 bytes for local variables. Maximum is 2048 bytes.
`

franksterke, 2021-11-17:
Same problem here, @CobaltEcho did you find a solution?

drf5n, 2022-04-15:
Those are just warnings, not errors.

They are a known issue in https://github.com/gnea/grbl/issues/228

You can work around them in a couple ways, per https://github.com/gnea/grbl/issues/228#issuecomment-1100454342

My preferred way is to use https://github.com/gnea/grbl/wiki/Compiling-Grbl#build-as-a-normal-arduino-ide-project 

I use this procedure for a grbl install :

1. From the command line:

        git clone git@github.com:gnea/grbl.git   # download grbl
        touch grbl/grbl/grbl.ino        # create an empty grbl.ino file for the IDE

2. Open Arduino IDE
3. File/Open/  (navigate to the grbl.ino file and open it)
4. Compile and configure in the IDE