gnea/grbl #1053: XLoader link on "Flashing Grbl to an Arduino" leads to compromised website

frazierjason, 2021-07-10:
The following Wiki page:
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

has a Windows instruction section that references a utility download called XLoader.  The URL referenced there leads to a website that (for me)  redirected to what looks like a fake Microsoft Security web portal. On reload, it later redirected to to a "domain for sale" page.  VirusTotal thinks it's clean, with one suspicious hit, but nonetheless this documentation URL no longer provides XLoader and should be fixed or replaced with something more suitable.
https://www.virustotal.com/gui/url/ad88560ad52a37573a1d0939af523f3bd39d758b5ad19016dffbb915e1f79729/detection

harryberlin, 2021-07-10:
check this:
https://github.com/zkemble/AVRDUDESS

PicEngraver, 2021-07-11:
XLoader is here on github:
https://github.com/xinabox/xLoader

PicEngraver, 2021-07-11:
Try link now.  Edited it to get XLoader from different source.

per1234, 2022-11-08:
Malicious edits were made to the Arduino IDE download links in addition to the XLoader links.

I did a full survey of the wiki edit history:

- https://github.com/gnea/grbl/wiki/_compare/614bc101b45abf181facea8e5cdd0596b6f576e8
- https://github.com/gnea/grbl/wiki/_compare/971f5fa9cfe133911c2b3144c20ae929de353ec6
- https://github.com/gnea/grbl/wiki/_compare/b6bda149181bec73a13d0f976d3b7eada50cac31
- https://github.com/gnea/grbl/wiki/_compare/deaeac6009be1f84c18fa76cd0f58144f0e0fbb6
- https://github.com/gnea/grbl/wiki/_compare/bf8da25354e311faaafc15ffae4819c59b3b7b2f
- https://github.com/gnea/grbl/wiki/_compare/ba2996fe53837bb357327f83ea7a1cd51c84e0fe
- https://github.com/gnea/grbl/wiki/_compare/2bbd5f8ef79e598bbd251bd4064fc27c8a4197d9

I reverted the malicious links back to the original intended ones.

An analysis of the general tactic:

https://www.crowdstrike.com/blog/how-threat-actors-use-github-repositories-to-deploy-malware/

I support making wikis publicly editable, but it really is important for people to carefully monitor the wiki if you are going to do that. From my experience doing that for the [`arduino/Arduino` wiki](https://github.com/arduino/Arduino/wiki), the number of malicious edits are small so it is not a huge task to revert them.

per1234, 2023-03-23:
Just adding an update to note that the malicious edits continue:

- https://github.com/gnea/grbl/wiki/_compare/3bf25469ec3e3b7502e86452c9d07b6df9d316f8
- https://github.com/gnea/grbl/wiki/_compare/e75c43deb8d4b8726d145525c677ef51e8c70e4a
- https://github.com/gnea/grbl/wiki/_compare/457e5c32e654082f77d3cc0404496c7c9f72fa99
- https://github.com/gnea/grbl/wiki/_compare/dba11e1ae27cc0eac38e2042fbb9458bcadd6f69
- https://github.com/gnea/grbl/wiki/_compare/f3cdeb6df325729eba45967f22c280db85b07d30
- https://github.com/gnea/grbl/wiki/_compare/77966776f84be8bfa01518446f123a8c23ea0815
- https://github.com/gnea/grbl/wiki/_compare/9a530b6fc1f1d23b64c502e5e6e24acd9f171216

All but one of the subverted links had already been fixed. I reverted the remaining malicious edit.