gnea/grbl #1056: Adding gbrl zip to the library

cmg360, 2021-07-19:
I have been trying to place gbrl code into the IDE without success.  Each time I try the I get the message that there is no valid library in the .zip file.  The file is downloaded from git-hub.  I have the latest IDE loaded and have not previously had gbrl loaded.  I have followed the directions and tried both importing .zip and a manual load but all result in the same error message. What am I doing wrong?

jrfantom, 2021-08-28:
i have the same problem, can anyone help

cmg360, 2021-08-29:
The action sent to me was to copy the files in but this did not create a library and I keep getting error message that there is a problem with the attempted GBRL library.

closed 2021-08-29 by cmg360