gnea/grbl #1059: Jerky processing of the G code

treki1, 2021-07-27:
Woodpecker controller board

Suddenly I have the problem that the G-Code only runs jerkily.
See attached video.
Does anyone know this problem?

I installed bCNC on a different laptop.
I flashed v1.1h.
Still the same thing :(

https://user-images.githubusercontent.com/11616336/127201639-757a2da0-6fbd-4c96-833b-2f7f0a88a9f3.mp4

PicEngraver, 2021-07-28:
Video appears to show a normal peck drilling cycle.   Is this what you are calling jerky?  Is it jerky with any gcode files other than drilling?

treki1, 2021-07-28:
Yes. Even with milling cycles. Spindle on-off-on-off and also the axes.

PicEngraver, 2021-07-28:
Hmmm.  Seems odd (but not impossible I suppose) to be something from a controller???  Maybe the gcode?  Can you post the gcode file used in the video?

treki1, 2021-07-28:
Yes. Here the attachement.
[1.bohren.1mm.ngc.zip](https://github.com/gnea/grbl/files/6893704/1.bohren.1mm.ngc.zip)

PicEngraver, 2021-07-28:
Thanks.  Gcode looks good.  A straight forward drilling cycle (not pecking), and nothing to suggest a problem.  Perhaps you have some bad cable connections?  
[image]

treki1, 2021-07-28:
I have already tested with a different USB cable. It made no difference.
The homing also works normally. Moving by hand also works normally.

PicEngraver, 2021-07-28:
Why do you have a G94 in your gcode?  This is unusual.  I think it is usually used with lathes.  Test without it.

Norwegian-Gadgetman, 2021-07-29:
G94 is not 'Lathe only', it just means 'Units of feed rate per minute'
Now, G95 is 'units of feed rate per revolution' and that's only really usd on lathes.
That G94 is not commonly used outside a lathe is another thing entirely.

The code also contains G21 which is 'mm distance units', and when that is used, feed speeds are assumed per second.
On G20 its inches and inches per minute.

There's two 'Feed Speed' commands in the code.
F54 just before a couple of Fast Move commands.

On older versions of GRBL this only sets the G1 speed.(the G0 would still be limited only by the machine limits) I'm not too familiar with the latest editions.
The second, F120 is either 120mm/s or 120"/minute. Take your pick.
It's way too fast for drilling anyways.

If F-commands also limits the G0 traverse speed, I would suggest adding a Fxx option to each G1 command.

G1 Z-3.5 F10
Gets you 10mm /second drill speed but only for that one command.

Trygve

treki1, 2021-07-29:
My G-Code is generated with Target3001 and has worked a few hundred times so far ;)
I have now ordered a new controller.

RabbitMountain, 2021-08-03:
I honestly didn't notice anything unusual from the video. Perhaps film another?

My only suggestion would be check your cables (not your USB cable). Maybe use shielded cables or add decoupling capacitors to reduce any noise?

I highly doubt you need a new controller board, but it seems you bought one anyway. If it is your cables or code you will have the same issue with your new board.

treki1, 2021-08-03:
The jerking also happens with older G-Codes, which ran smoothly at the time.
The jerking also occurs when routing the conductor tracks.
I will then inform you.

treki1, 2021-08-21:
Ich warte immer noch auf den neuen Controller.
Habe mittlerweilen herausgefunden, dass wenn ich beim Spindel-Start "Pause" drücke und vorfahre, dann alles normal läuft bis zum nächsten Werkzeugwechsel.
Ich bin sicher, der Controller hat ein Problem. Habe ja oben geschrieben, dass ich bCNC auf einen anderen Computer installiert habe zum test. ...genau der gleiche Fehler!!

I'm still waiting for the new controller.
In the meantime I have found out that if I press "Pause" when starting the spindle and drive up, then everything runs normally until the next tool change.
I am sure the controller has a problem. I wrote above that I installed bCNC on another computer for the test. ... exactly the same mistake !!

treki1, 2021-08-21:
New controller & spindle motor ordered.
The new motor is 20000 1/min @24V. Old 10000 1/min @24V (12000 @36V).
I make a upgrade on my cheap CN milling machine ^^

treki1, 2021-08-21:
M03 S1000
G4 P 0,5
M03 S2000
G4 P 0,5
M03 S5000
G4 P 0,5
M03 S10000
will solve the current problem with the 5.5A supplie.