gnea/grbl #1061: Status Reporting while homing

jaimealemany, 2021-08-18:
Hello,

I'm writing an automated interface to grbl, and I'm finding that the ? character is not being paid attention while the machine is homing.

After the homing is completed, I get a single reply (whatever the number of ? I sent): <Home|...>
After that, next ? return: <Idle|...>

Is this the intended behavior? I understood ? should always immediately reply back.

(my device has currently version 1.1g)

Thanks.

langwadt, 2021-08-18:
afaik it has always been like that, possibly originated in an oversight, but then left as-is to not change the behavior

jaimealemany, 2021-08-18:
Thanks for replying. I just added a short note in the "Interfacing" section of the wiki.

closed 2021-08-19 by jaimealemany