gnea/grbl #1069: Change pin configuration in a custom atmega328p board

andrevargas22, 2021-09-08:
Hi there, 

So, I've made a custom board with a standalone atmega328p, running with an Arduino bootloader (it might not be the best option, but that's how I know to run it). And essentially everything is working as planned. I tried to route the pins in my board exactly the same as they are defined in GRBL so I wouldn't have problems and my plan worked, stepper motors are working, hard limits are working. So far so good, the world is beautiful. I'm using GRBL v1.1.

However, I've made a mistake in routing and I need to change one pin in order to finish my project. And it is, from cpu_map.h:

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_DDR   DDRC
#define COOLANT_FLOOD_PORT  PORTC
#define COOLANT_FLOOD_BIT   3  // Uno Analog Pin 3

In my project, the coolant is assigned to PORTC1 (Uno Analog Pin 1), so my logical assumption was that I should just change the bit from 3 to 1, like this:

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_DDR   DDRC
#define COOLANT_FLOOD_PORT  PORTC
#define COOLANT_FLOOD_BIT   1

And after compiling GRBL, this pin starts in a HIGH position, as it should, but when I send the command M8 to change the pin status to LOW, GRBL resets instantly and gives me [MSG: Check limits].

If I compile the code as PIN 3, the microcontroller works as supposed to, M8 and M9 change PORTC3 status from HIGH to LOW and vice versa with no errors.

Does anyone have a clue why? By using PORTC1 am I changing something that I shouldn't? Thanks in advance.

langwadt, 2021-09-08:
check the other defines to see if that pin is already used for something else

andrevargas22, 2021-09-08:
I can't believe I'm this blind. That was so obvious I'm almost ashamed of myself.

Yes, the pin was being used somewhere else, I changed it to an unused pin and now PORTC1 is working as I wanted.

Thank you langwadt, I'm closing this issue.

closed 2021-09-08 by andrevargas22