gnea/grbl #1070: GRBL Post Processor Failing in Fusion 360

rhdhybrid, 2021-09-09:
I know, probably the wrong place to ask, but has anyone had issues with the GRBL Post Processor in Fusion 360, Everything worked in June, tried to make a Mickey Mouse Sign for Niece and it fails every time, Nothing changed in my machine profiles or in Xpro V5. Seems to be a issue in fusion it self. I wrote Autodesk and anyone else I can about it but can't seem to find a answer, the same tool path will post process using the Maslow CNC Post Processor so I think it is the GRBL one that is the issue. Thanks again, if this isn't the place then I'm sorry. 

Ken

bstein2379, 2021-09-10:
It's been working for me. I'm relatively new to CNC, but I've been using the Fusion 360 Grbl/grbl post processor for the last few weeks. One thing I had to change was the Safe Retracts setting from G28 to Clearance Height to keep my end mill from plunging and then running off into the work surface at the start of the carve. That was scary for a noob like me.

From what you describe though, it seems like you have a different issue.  Does the actual posting of the nc file fail after you click the Post button? What error does it show?

BernardG, 2021-09-10:
Could you define "failing"? What, EXACTLY, is the problem?

rhdhybrid, 2021-09-10:
#Sorry for the lack of description. So the issue is that the fusion 360 grbl post processor does not support the mist coolant using the m7 command. If you change everything back to just flood coolant it will work and post correctly using the M8 command. With a-bit of manual editing it worked fine, changed M8 to M7 and way to go. However I think the current grbl post processor doesn't support the current commands available in the Cnc xpro v5 controller. Perhaps the xpro v5 needs it own post processor but that's way over my head. Lol

GrahamJB, 2021-09-10:
Isn't that a Fusion360 problem rather than a Grbl problem? You may get better answers from F360 forums.

neilferreri, 2021-09-10:
Should be an easy fix in the post processor.
Change 
{id: COOLANT_MIST},
to
`{id: COOLANT_MIST, on: 7},`