gnea/grbl #1071: Ignore endstops after homing cycle

davidfobar, 2021-09-14:
I am using GRBL for something very different than a CNC application, and after I home the system I would like to ignore a specific homing switch and only rely on software limits. I cannot seem to figure out the correct set of settings. Is this something that needs to be changed in the source code, and if so in which file?

For more details, I am rotating a table and it needs to be able to rotate 380 degrees, currently, once I reach about 355 degrees I encounter my limit switch and it causes the whole thing to stop the operation.

Thanks

PicEngraver, 2021-09-15:
Disable the Hard Limits (set $21=0), but leave Homing enabled ($22=1).

davidfobar, 2021-09-24:
Thank you, problem solved. I was attempting to send the command with the wrong format. Thank you for your assistance.

closed 2021-09-24 by davidfobar