gnea/grbl #1072: Is there an upper limit for steps/mm to behave reliably?

byoungUST, 2021-09-14:
I've run into an issue with using GRBL in a special case.  I am using a serial connection through the Arduino IDE and a MKS DLC V2.1 board.  
The 3 stepper motors are independent of the motion of the other motors.  I have only altered the X/Y/Z steps/mm and X/Y/Z maximum speed.

I have been using the TMC2225 stepper driver for X and Z and they worked fine.  I tried using a TMC2209 for the Y Motor due to the higher current load, the problem arose when I tested the Y axis motor to confirm the configuration.  
GRBL's status command comes back garbled and the distance covered by Y was not correct.

The TMC22xx series have a 256 micro step interpolator that is active by default and none of the step pins have any affect on that functionality and so I compensate for that with GRBL's configuration to get the proper distance.

For the TMC2225 and TMC2209 the steps/mm were set as shown below:
| Config $x       |  value             |
|----------------|----------------|
|	$100	|	814.871	|
|	$101	|    40314.960 	|
|	$102	|      1007.874	|
|	$110	|	6000.00	|
|	$111	|	6000.00	|
|	$112	|	6000.00	| 

To check and see if the steps/mm value had some upper limit for reliable operation of GRBL  that I exceeded  I switched to a DRV8825 chip. A chip that didn't do anything fancy and paid attention to what the micro step pins were set to.

The settings for TMC2225 on X and Z and DRV8825 on Y are:
| Config $x       |  value             |
|----------------|----------------|
|	$100	|	814.871	|
|	$101	|  	314.960	|
|	$102	|      1007.874	|
|	$110	|	6000.00	|
|	$111	|	6000.00	|
|	$112	|	6000.00	|

The new setting works as intended so I'm wondering there is an upper limit for the steps/mm to have GRBL operate reliably.  I don't expect this to be fixed as the problem arose from a very special case and I've found functional work around.  I just want an explanation and what the safe upper limit value is for steps/mm.

langwadt, 2021-09-14:
comments in the code says; step/mm * mm/min * 1/60  should stay below  ~30kHz, 

1007.874 * 6000/60  = ~100kHz ... too fast or too many steps/mm  
314.960 * 6000/60  = ~ 31kHz so right on the limit

byoungUST, 2021-09-14:
1007.874 works fine actually.  It's Y that was the issue.  Thanks for the answer though.

closed 2021-09-14 by byoungUST