gnea/grbl #1074: How to go back to grbl0.8c?

wanabegeek, 2021-09-17:
Hi all. I'm pretty new to this. Bought a cheap laser engraver from a chinese website. It came with no instructions, no specs etc. I got it built and tried to do some tests but it never worked properly. I realise now, after researching here, the problem was probably in a setting I had overlooked. What did work was the laser. It only came on when a job started, although it would burn the initial travel line from home to the start of the design.

In the course of trying to get it to work I found out that I've got an Arduino Nano. It came with grbl 0.8c. After reading lots of info here, I decided to flash grbl 1.1h. Whilst it gave me more control options I cannot turn the laser off. I've adjusted the spindle settings $30 =(was 1000,  changed to 255), $31 =0, $32=0 (have tried both on and off). I've edited the config.h file to comment out #define VARIABLE_SPINDLE. And still I have no control over the laser power. I've obviously tried using the M3-M5 commands through lasergrbl but they have no effect. 

So now I think my only option is to go back, flash grbl 0.8c back onto the Arduino. But I can't find the files anywhere. Any help or advice would be much appreciated. 
Thanks

SimonJohnCastle, 2021-09-17:
Did you use S0, you don't mention it, only M3-M5

langwadt, 2021-09-17:
afair spindle enable and spindle pwm was moved to different pins after 0.8. You could try to modify the code or hw to use 1.1

wanabegeek, 2021-09-17:
S0 also does nothing unfortunately

PicEngraver, 2021-09-17:
[grbl_v0_8c_atmega328p_16mhz_9600.zip](https://github.com/gnea/grbl/files/7186791/grbl_v0_8c_atmega328p_16mhz_9600.zip)

langwadt, 2021-09-17:
or if 9600 is too slow, rebuild https://github.com/grbl/grbl/tree/v0_8

wanabegeek, 2021-09-18:
Langwadt you are a star! Thank you so much. Flashed 0.8c, made the necessary changes to the settings, and now I have a working laser engraver for the first time.

closed 2021-09-18 by wanabegeek