gnea/grbl #1075: Optocoupler  wiring need help

Kamoba, 2021-09-17:
I'm using a transistor output optocoupler PC817

[image]

But it's not working before i added the the wiring showed [here](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches) all was working with a lot of noise. But now limits switch are not detected.

[image]

**My settings for NC:**

> $0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)

What i did wrong?

SMTDesigns, 2021-12-14:
your 1K resistors on the output of the opto-coupler are too low. Change to 10K but not lower than 4.7K.  You have approx 20ma for forward current on the LED. The collector current should be 1ma instead of 5ma that you have. The spec sheet doesn't have a graph for collector current related to saturation voltage of the transistor. It does spec 1ma with a 0.2 volt max value.

Kamoba, 2021-12-14:
> your 1K resistors on the output of the opto-coupler are too low. Change to 10K but not lower than 4.7K. You have approx 20ma for forward current on the LED. The collector current should be 1ma [...]

Thanks SMTDigns, i already move forward with other schematic (resistors + capacitors) because this cause to me a lot headaches and finally a short-cut on my pi... But I'll try that next time.