gnea/grbl #1083: More Information in Hard Limit Error

anotherboringuser, 2021-10-04:
Hi, 
I’m currently troubleshooting problems with my limit switches on my cnc machine. I’m getting this error for no apparent reason: 

> [Error] An error was detected while sending 'X20Y-21Z-0.6I0J-1': (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended. Streaming has been paused. 

Now it would be very very useful if GRBL also told me **which** limit switch triggered. This would help me narrow down the problem. Situations where this might help are for example NC switches connected through broken wires. 
Am I missing something and this feature already exists? Or is there a reason why it doesn’t?

langwadt, 2021-10-04:
there's single interrupt for pin change, if the signal is gone before the interrupt routine can read the pins there is no way of telling which one caused it

GrahamJB, 2021-10-04:
And the switches are wired in Parallel on the pins, both limit switches on an axis look the same so it could not tell you if it was the + or - switch that had been trigered. The best way to find out which switch has triggered is to look at the machine or get switches with an inbuilt LED which lights when the switch is triggered.

anotherboringuser, 2021-10-04:
> And the switches are wired in Parallel on the pins, both limit switches on an axis look the same so it could not tell you if it was the + or - switch that had been trigered. The best way to find out [...]

Narrowing it down to one axis would already help a lot. I have LEDs on the switches that show what my limit switches are doing, but unfortunately they don't show EMF issues.

anotherboringuser, 2021-10-04:
> 
> 
> there's single interrupt for pin change, if the signal is gone before the interrupt routine can read the pins there is no way of telling which one caused it

Is there a way to change that or another way to see which pin is responsible for the triggering? Or would that add too much additional complexity to GRBL?

langwadt, 2021-10-04:
@anotherboringuser The interrupt fires if any of the pins changes state, there is no way to tell which one other than reading the pins, but the signal might not be there anymore for example because it was just a short glitch.

SimonJohnCastle, 2021-10-05:
Looks like its time for a limit switch buffer project with latching output and trigger indication.
This would be very simple to implement with an additional Arduino.