gnea/grbl #1093: Problem with XYZ core laser machine

marcootaviomello, 2021-10-31:
My problem is not LaserGRBL related, it's just a machine.

I recently had to move the laser module on some machines. She is an XYZ core machine.
When I went to test, I have a problem that the photos will be able to analyze better, when or project, move you do
engines, seems to have a loss of synchronism.

As an example, I tested how a square, or you, submitted a move twice at a time, and got a satisfactory result.
Therefore, when undoing a circle, other images that have curves, I get or result in an image.

I'm an initiator and I don't even know, it's a mechanical problem, which I don't think it is, to find out in a machine, or it could be
something in the settings.

Sorry to stray a bit or focus on two forums, but I thought I might find help here.

Thank you very much.
[image]
[image]

PicEngraver, 2021-11-02:
Circle is showing classic axis backlash issues (from loose belt?).  Check your belt tensions.

Host1992, 2021-12-29:
I have the same issue and I can't find anything about it 
[image]