gnea/grbl #1094: Stationary tool probe question

xeno-kalo, 2021-11-08:
Hello everyone,

I recently purchased a tool probe which inside the housing contains a basic switch (can be wired NO or NC).
I have to tool probe bolted down on my machine bed's top right hand corner.
I use GRBL and UGS to control the machine.

I am trying to use the probe to help me semi-automate tool changes.
Can any one you confirm if the below workflow is achievable with GRBL + UGS combination ?

1. Manually Load Tool 0 in the spindle
2. Manually set Z=0 on my stock
3. Probe the Z axis using my tool probe
4. Manually Load Tool 1 in the spindle
5. Probe the Z axis again
6. Calculate the difference in height between the 2 tools used

No matter how hard I tried I couldn't come up with a method to exclusively achieve the above using just WCS's...
Any tips? is this even possible to do? or there has to be a mathematical operation for this workflow to become feasible ?

All the best
Xeno

GrahamJB, 2021-11-08:
You will find it easier if you split it into 2 files but don't change the XY origin in between.

xeno-kalo, 2021-11-08:
@GrahamJB not if the stick-out length of the 2 tools is not the same :)

GrahamJB, 2021-11-08:
Are you talking about G43.1? That is the only thing I can see supported in Grbl which could be stickout.

I see no reason to use it, once you have set the Z zero using a probe with the bit tip then the tip of the bit is at Z zero regardless of what the stickout is. If the probe tool you refer to is just a contact switch that should work, if not I am not sure what you are using or trying to achieve.

terjeio, 2021-11-09:
I do not know if UGS supports tool changes with touch off for setting the tool offset in the same program, Grbl by itself certainly does not. I know since I have implemented a tool change protocol extension in my 32-bit port.
Better to ask at the UGS repo?

MARIOBASZ, 2021-11-09:
I don't know if ugs has a tool change macro implemented. bcnc has implemented.

xeno-kalo, 2021-11-10:
Thanks guys. I gather from the comments that indeed UGS cannot support this functionality. Bummer, and I liked the simplicity of it :D

closed 2021-11-13 by xeno-kalo