gnea/grbl #1096: grbl crash only when laser fires

rsaylr1, 2021-11-14:
hello,
I'm trying to find a working solution for my k40 setup.

omtech red k40
myjg-60w lpsu
omron s8vs 24024 psu
arduino uno clone
cnc shield v3 protoneer
grbl 1.1H
Lightburn

currently i have everything working properly, almost. The machine homes correctly, jogs, It will run "dry tests" from lightburn. By that i mean it will go through all the motions and go back to home as long as the laser is NOT enabled. If the laser is enabled, it goes to location and starts firing but stops within 5 seconds. at that point the arduino uno usually needs to be reset. im using z+ limit pin D11 for pwm and that works. I know this because I had grbl .9j loaded and everything run fine. only problem with grbl .9j is there is no laser mode $32  so I am getting burnt corners.
If i disconnect the laser pwm wires from the shield (D11) and run some gcode with the test fire button held on, the laser fires and the code completes as normal. I assume I'm getting some interference on the laser fire wire since the code runs with the laser off, or laser on but disconnected from the shield. it is just weird that if i only change from grbl 1.1H to grbl .9J the issue goes away so that makes me think there may be something else going on.

any suggestions?
 I'd like to put a filter of some kind on the laser fire wire (D11) but I don't know how and can't seem to find that anywhere.

Thank you for taking the time to read and help a guy out.

hairu, 2021-11-14:
You may be picking up EMI on the laser fire cable.  If you are able to, try shortening the cable and using a properly shielded cable (with the metal jacket connected to ground on one end).  If that doesn't work you can try using an optocoupler if the input is not already optically isolated.

rsaylr1, 2021-11-14:
That is on my short list to try. Any thoughts on why it doesn't happen when running grbl .9? Everything is exactly the same.. just swapping firmware...

rsaylr1, 2021-11-15:
Do you know if this (NOYITO 4-Channel Optocoupler Photoelectric Isolator Module Level Voltage Converter Module PLC Signal Converter Module PNP NPN to NPN (5V to 12V) ) would support pwm function? Does any optocoupler support pwm? This is out of my level of understanding...
Thanks

terjeio, 2021-11-16:
@rsaylr1 You should first check if v1.1 leaves any input floating, if so connect it/them to ground or 5V.

Optocouplers can be used for PWM, but depending on the PMW frequency you may have get a higher specced one. It is also important to load it correctly, to little diode current and/or collector current will degrade the switching timing quite a bit.

rsaylr1, 2021-11-21:
Just wanted to give an update. I purchased (noyito 4_channel optocoupler 5v to 5v) from amazon and this fixed my issue. I can also say that the 5v to 5v opto responds to the 0-5v pwm signal from my arduino uno board. it seems to scale just as if it was a direct connection. i was pleasantly surprised.
thank you for your help.