gnea/grbl #1101: Branch of grbl which supports plasma cutting and THC ?

bLouChip, 2021-12-03:
Has anyone considered a branch of this grbl project which supports CNC plasma cutting w grbl managed Z axis Torch Height Control (THC) ?  

Where by THC Z axis control is driven by grbl with inputs from an external THC unit that monitors the arc voltage.  There are 3 grbl hardware input pins from the external THC arc monitor which are: arc OK, AxisPlus, AxisMinus.  An example external THC unit is the Proma THC 150, https://www.proma-elektronika.com/compact-thc-150/
A new gcode for spindle control would be necessary, a variant of M3, for example: 
M3.1 <axis>, <axis_speed>, <axis_initial_delay>, <axis_max_dist>

GRBL function upon processing the M3.1 command:
a) turn on spindle and begin to monitor for input pin ArcOK as interrupt _while continuing to process remaining gcode commands in the job file and until an M5 is processed._  The function points b, c, d, are input pin and timer interrupt driven. 
b) upon sensing ArcOK, set timer for <axis_initial_delay>. 
c) When timer <axis_initial_delay> expires and while ArcOK remains present, begin monitoring for input pins AxisPlus and AxisMinus as interrupt.
d) upon sensing either AxisPlus or AxisMinus and while ArcOK remains present, move <axis> at <axis_speed> in the direction of either AxisPlus or AxisMinus for maximum distance of <axis_max_dist> +/- the position of <axis> at time of M3.1.   <axis_speed> (in units/min) and <axis_max_dist> are given in units mode (G20/G21) at time of M3.1 command. <axis_max_dist> is always relative from position of axis at time of M3.1.

If implementing grbl on Arduino Uno with limited available input pins, consider multi-tasking XYZ homing input pins for ArcOK, AxisPlus, and AxisMinus.  If implementing for mega256 chip, there should be ample available input pins.  

Thanks for consideration.

terjeio, 2021-12-03:
I started [this plugin](https://github.com/grblHAL/Plugin_plasma) some time ago for my 32bit port, but as of now it is untested as I do not have a machine to test it with.

bLouChip, 2021-12-05:
tergeio -  very nice plugin.   That's much more function than what I envisioned, but I get the gist it.  I'd be interested in being a tester however I don't have the hardware that supports grblHAL.  If I make such plans, I'll be in touch.  My plasma CNC solution has been Mega2560 controller and an external Proma THC SD, which requires re-homing the Z axis after each job. 

Anyway, good to know there is some grbl code out there that manages THC function.  Well done.

closed 2021-12-05 by bLouChip

cri-s, 2021-12-06:
Do you have "arc on" feedback from
Plasma ?