gnea/grbl #1107: Axis travel resolution - unpredictable behavior

T-vK, 2021-12-30:
I'd like to calibrate my 3018 CNC, but any attempts to teach it how many steps equal a mm of travel distance on an axis fail.
I'm using the latest firmware release (1.1h) and I've tried setting `$100` to a variety of different values, but non of them yielded the required results. 

Shouldn't a higher value result in more travel distance on the axis and a lower value in less travel distance?
It behaves extremely odd for me:

```
500 = 32.5mm
300 = 24mm
295 = 3.5mm
290 = 3.5mm
285 = 3.5mm
280 = 3.5mm
275 = 6mm
260 = 5.5mm
250 = 3mm
200 = 2.5mm
90  = 20mm
85  = 1mm
80  = 8.5mm
60  = 0.8mm
55  = 0.8mm
50 = 73mm
30 = 90mm
```
(On the left is the value I set $100 to and on the right is the physical distance the spindle moved on the x-axis.
I used UGS with a step size of 10 and a feed rate of 686 in the UI for every measurement.)

My settings:
```
Grbl 1.1h ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 2    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 1500.000    (Homing locate feed rate, mm/min)
$25 = 1500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 1500.000    (X-axis maximum rate, mm/min)
$111 = 1500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 295.000    (X-axis maximum travel, millimeters)
$131 = 150.000    (Y-axis maximum travel, millimeters)
$132 = 7.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
```

About my hardware:
The stepper motors (17HS3401) do 1.8° per step (200 steps per rotation).
The threaded rod moves the spindle by roughly 4mm per rotation.
The drivers are A4988. (I think at a 1/16-step microstep resolution.)
So theoretically I would have to set $100 to 200*16/4=800 ... right?

PicEngraver, 2021-12-31:
My not knowing your hardware setup, I'm just guessing that it may be that you are missing steps.  Does the zero position change?
Try reducing settings $110 and $111 to 1000 and retry.

T-vK, 2022-01-01:
Thank you so much, PicEngraver! Reducing the $100 and $111 values totally fixed my issues. 
My calculated value of 800 for $100 works like a charm now. :)

PicEngraver, 2022-01-01:
Have a Happy New Year.

T-vK, 2022-01-01:
Thanks, you too! :)

SimonJohnCastle, 2022-01-02:
My CNC3018 came with a woodpecker 3.4 board, I also had problems with missing steps in the beginnig.
Make sure you put some lythium grease on the threads and bearings.
Mine works fine at max 2250 which is fine for jogging around, I did not configure the pots but if you find its not fast enough
try increasing the driver current. The 3.4 board is OK but doesnt have any microstep adjustment which would be nice to go faster at the expence of accuracy.