gnea/grbl #1108: M226 Gcode command implementation

erosalesodev, 2022-01-02:
Hi, am trying to implement the **M226** wait for pin command, so far I have a solution that is nothing more than a simple `while` that read the state of the pin, the problem with that is the cycle is blocking, in case other commands arrive at the moment it's executing will not be processed, someone has ideas how to solve this.Thank you for read.

terjeio, 2022-01-02:
You may call `protocol_execute_realtime()` or [delay_sec()](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/nuts_bolts.c#L111-L126) with appropriate parameters in the loop (and perhaps exit it on `sys.abort`). But it will still be blocking new gcode and $ commands - which is the intention?

An alternative to M226 would be to implement [M66](http://www.linuxcnc.org/docs/html/gcode/m-code.html#mcode:m66) or use `M0` (after a `G4` to finish queued motions) which will wait on the cycle start input.

erosalesodev, 2022-01-02:
Yes i have tried this
```
 system_set_exec_state_flag(EXEC_FEED_HOLD); // Use feed hold for program pause.
   protocol_execute_realtime(); // Execute suspend.

  while(digitalRead(pinVal)!= stateVal){
   
  }
  printPgmString(PSTR("Pin ready\n"));
  system_set_exec_state_flag(EXEC_CYCLE_START); 
  protocol_execute_realtime(); // Execute.
  
```
`pinVal` is the pin number and `stateVal` is the logical value
This is used in plasma cutting to wait for an Arc Ok signal that indicates that the cut voltage is stabilized, it is necessary to wait for it to start.
This is why wait for pin is implemented,stop the movement works but still having problems with restarting the movement cycle.
The main idea is try to implement the waint for pin command that allows waint for the Arc Ok signal and stop the movement until appears, and the restart the movement with all commands, the stoped queue and the recived in the waint

erosalesodev, 2022-01-02:
Thanks @terjeio for the idea i will try `sys.abort` and `protocol_exec` functions

MARIOBASZ, 2022-01-02:
How is the Pin chosen to use M66 declared?

erosalesodev, 2022-01-05:
Done

```
protocol_execute_realtime(); // Execute suspend.
  while(digitalRead(pinVal)!= stateVal){

  }
  protocol_exec_rt_system();  // Executes run-time command`
```

closed 2022-01-05 by erosalesodev