gnea/grbl #1111: Hard Limit Alarm constantly set off

masssassin06, 2022-01-31:
I'm using NC switches for my hard limit setup and an Arduino uno. The machine is self-made. Currently when connected, but with the machine(motors) switched off, the limits work as required setting off the alarm when pressed; however as soon as the system is powered on Alarm 1 is triggered as though one of the limits has been pressed even after resetting the Arduino. When checking with a multimeter, the voltage at each of the limit pins is still held high. Any help as to what could cause this issue or what could solve it?

MarkEMark2, 2022-02-12:
I logged in to ask the exact same question. This could have been my original post as well! 

Attached is a screen shot of the alarm and the GRBL settings as revealed in UGS. I see this was asked 13 days ago... Anyone know what we may be facing here? All motors and axis work when running the setup wizard, and I have successfully calibrated the movement. But as soon as I enable the homing and limit switches, I get the Alarm 1 limit switch alarm. All the switches are checked (they are NC switches) and all are functioning and wired correctly. I traced the wiring all the way to the circuit board and all of the wiring is functional. The switches work, but the Z axis is flagging an immediate error upon enabling them. Triggering x and y work as planned. And, as you see in the picture, the x and y axis are green with the same wiring set up as the z axis. 

Attached are the config.h file as I have attempted to look there for answers and have made the changes along the way when things didn't work on first blush. You can see my GRBL settings in the graphic attached as well.  Any help for this would be greatly appreciated as the machine is finally ready to use, but the software set up is throwing this error. I am using:

- HobbyCNC 3 axis driver - new board.
- GRBL 1.1
- Arduino Uno R3 (also tried the Mega 2650 first and had the same issue
- Latest version of UGS and GrblGru (both seeing the same error)
- I installed a limit switch filter in line with the switches to minimize noise. I have wondered if there may be an issue with this board as the wiring all tests positive when checking them from the switches to this Opto board, and from the opto board to the driver board. All wires are good. power to the opto board and to the driver board on both sides of the opto couplers tests 5 volts.

In summary, all motors work and control properly until I try to use the limit switches, THEN I get an immediate z axis alarm saying I hit a limit switch - but none are active.

Thanks for the help!

[config.h.txt](https://github.com/gnea/grbl/files/8054688/config.h.txt)
[image]
[image]

masssassin06, 2022-02-25:
> 

Hey, so I actually ended up being able to work through the issue and it turns out my microcontroller (Arduino uno) was picking up interference from the Motor drivers. A simple Low-pass filter ended up being able to solve the issue. Another way would be to increase the distance between the controller and the drivers. Hope this helps.