gnea/grbl #1115: How to ensure GRBL is really IDLE?

naikymen, 2022-02-16:
I was wondering what is the best way to ensure that the machine has stopped moving.

The objective is to ensure an idle machine, before a different machine will perform an independent but coordinated action (which requires all GRBL moves to complete, and an idle machine).

So far:

* I was able to count 'ok' responses (which do not ensure the end of movements, but was a first step) of every move gcode.
* I was also able to monitor GRBL's status, by continuously sending '?' commands and parsing the outputs.

Because receiving all 'ok' responses ensures that the serial interface is not messing with me, I had imagined that a subsequent "idle" status response would ensure the machine was not moving, and would not move afterwards.

However, though infrequently, the machine was not actually IDLE, and would continue to move.

I was wondering if there is a fool-proof way of ensuring that a machine has finished moving.

So far, my best strategy has been to require three consecutive IDLE status reports, instead of just one.

Thanks in advance!

langwadt, 2022-02-16:
afair a dwell command (something like G4 P0.01) won't reply 'ok' until the buffers are empty and grbl is idle

naikymen, 2022-02-16:
Thanks for the quick reply!

G4 commands seem adequate, I was unaware of them. Closing!

For anyone who failed to find it, a [list of supported commands](https://github.com/gnea/grbl#update-summary-for-v11) is in the main readme :)

closed 2022-02-16 by naikymen