gnea/grbl #1116: X limit Switch triggers XZY

NilsNippe, 2022-02-27:


Pethical, 2022-02-28:
Hi,
Please check your wiring.

NilsNippe, 2022-02-28:
Do you have a concrete wiring error to check? The switches are connected to the corresponding pins as referenced here: https://github.com/gnea/grbl/wiki/Connecting-Grbl with the modification of adding one of the y-axis limits in series with the z limit.

There are no shorts between the switches.

When the arduino reads the switches with a dedicated program that does nothing else, the limits work as expected.

NilsNippe, 2022-03-05:
In my case it was the typo in the report.c grbl file. https://github.com/gnea/grbl/issues/712
After I redownloaded all of the grbl files my router worked in single y axis mode, and compiling grbl for dual axis gave the corrosponding error.

closed 2022-03-05 by NilsNippe