gnea/grbl #1118: Yaxis movement triggers alarm first move after homing

i-g-g-y, 2022-03-04:
I have grbl 1.1h. Have X, Y and Z homing, VARIABLE_SPINDLE is disabled, Homing is performed successfully.  

After Home cycle I can move X and Z axis as I wish, but as soon as I move Y axis an ALARM is thrown. I do not know which, it does not say, but in the command line window it claims a hard limit was triggered, but this is completely false:

An unexpected error was detected: (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.
[MSG:Reset to continue]

This happens regardles of the direction of the move, -Y or +Y.

I tried flipping the Y axis, adjust homing and also tried movement in -Y and +Y. No matter what I do, it will throw an alarm at me.

What is wrong?

langwadt, 2022-03-04:
99% chance it is noise on the wires

i-g-g-y, 2022-03-04:
0% chance. I filtered the damn thing to such extent that the machine actually crashes during homing (no damage) and it still behaves exactly the same.

Verbose output:

[verbose] <Idle|WPos:-85.000,18.575,-15.000|FS:0,0>
>>> $J=G21G91Y20F100
ok
[verbose] <Jog|WPos:-85.000,18.600,-15.000|FS:72,0|WCO:-195.000,-155.000,-5.000>
An unexpected error was detected: (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.
[MSG:Reset to continue]

langwadt, 2022-03-04:
what happens if you put the machine middle of the work area and just reset without homing, can you then move Y?

i-g-g-y, 2022-03-04:
I can't move machine in either direction for y axis. It throws error.

nielsnl68, 2022-03-04:
Did you already tried to remove the end-stop cables one by one, while trying to move the yaxis?
I'm with Langwadt that it has could noise.  I had the same thing with my grblshield years ago.

How did you create that filter. is that within software or hardwere?

daxliniere, 2022-03-04:
I'm having the exact problem with mine and it's due to noise from the steppers.
Try to set your feed rate very slow. It will generate very little EMI and will probably not set off your limit switches. That's the way to know where your problem lies.

The other thing you can try is to simply jumper the limit switch pins at your GRBL board and see if your machine runs. If so, then it's definitely noise and probably from the steppers.

Good luck!
Dax Liniere.

daxliniere, 2022-03-04:
Also, have a read of this: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

i-g-g-y, 2022-03-04:
> Did you already tried to remove the end-stop cables one by one, while trying to move the yaxis? I'm with Langwadt that it has could noise. I had the same thing with my grblshield years ago.
> 
> How did you create that filter. is that within software or hardwere?

HW. Also duringnormal operation, the normally closed switches will not pick up a lot of noise, but sure, I will try tomorrow. Removing my cables means all my switches are pressed, I could find jumpers and try to short every thing on cnc shield.

i-g-g-y, 2022-03-04:
> Also, have a read of this: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

Read it, thanks.

i-g-g-y, 2022-03-04:
> I'm having the exact problem with mine and it's due to noise from the steppers. Try to set your feed rate very slow. It will generate very little EMI and will probably not set off your limit [...]

Final approach is 25mm/min, plenty slow, but thanks for suggestion.

i-g-g-y, 2022-03-05:
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 4    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 16    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 100.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 200    (Homing switch debounce delay, milliseconds)
$27 = 20.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 161.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 300.000    (X-axis maximum travel, millimeters)
$131 = 160.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok

i-g-g-y, 2022-03-05:
> 99% chance it is noise on the wires

100% chance. I did filter, but then decided to filter properly, so added capacitors on cnc shield instead of cables. Now it works.

closed 2022-03-05 by i-g-g-y