gnea/grbl #1121: SAFETY_DOOR_SPINDLE_DELAY 4.0

Nipote, 2022-03-11:
// After the safety door switch has been toggled and restored, this setting sets the power-up delay
// between restoring the spindle and coolant and resuming the cycle.
#define SAFETY_DOOR_SPINDLE_DELAY 4.0 // Float (seconds)

Hi everyone, can you tell me how I can change this parameter !? I need to change from 4 seconds to 10 seconds!!

Thanks Simone

Nipote, 2022-03-11:
0x9E : Toggle Spindle Stop

    Toggles spindle enable or disable state immediately, but only while in the HOLD state.
    The command is otherwise ignored, especially while in motion. This prevents accidental disabling during a job that can either destroy the part/machine or cause personal injury. Industrial machines handle the spindle stop override similarly.
    When motion restarts via cycle start, the last spindle state will be restored and wait 4.0 seconds (configurable) before resuming the tool path. This ensures the user doesn't forget to turn it back on.
    While disabled, spindle speed override values may still be altered and will be in effect once the spindle is re-enabled.
    If a safety door is opened, the DOOR state will supersede the spindle stop override, where it will manage the spindle re-energizing itself upon closing the door and resuming. The prior spindle stop override state is cleared and reset.

   "When motion restarts via cycle start, the last spindle state will be restored and wait 4.0 seconds (configurable) before resuming the tool path. This ensures the user doesn't forget to turn it back on."

MARIOBASZ, 2022-03-11:
Modify to
#define SAFETY_DOOR_SPINDLE_DELAY 10

Nipote, 2022-03-11:
Hi Mario, thanks for the answer, could you tell me something more... where should I write it?

I am using open OpenBuildControl... 
Do I need to use another program?

I read on GitHub that you can change that value but I don't know how and where!

MARIOBASZ, 2022-03-11:
inside the grbl file that you uploaded to Arduino, grbl file, config.h folder, I think on line 179 you will find it.

Nipote, 2022-03-11:
Hi Mario, I haven't uploaded the grbl to arduino yet, I just waded in how to do it but I don't understand how to modify it, could you guide me from the beginning, you can tell me all the steps. thank you!!
It seems to me a very difficult thing, I imagined it easier! Shouldn't it be a simple value to change according to the spindle you are using ??

Nipote, 2022-03-11:
"Shouldn't it be a simple value to change according to the spindle you are using??"
I mean by the simple G-code..

MARIOBASZ, 2022-03-11:
You are right, but it is so, it is not loaded to modifyvfrom a $number command.
1.0 -Do you have your arduino?
1.1- Do you have your grbl software to upload it to the Arduino?
If you have both:
2-You can save the soft on your arduino (if you can't save it, maybe you made a mistake in the location of the files), familiarize yourself with it.
If so far you go ok, to modify and view the files, I use the sublime text program. Find the file config.h , modify the line to change the timeout, save the modifications and. Burn the Arduino with the modified program.

Nipote, 2022-03-11:
I am using an Openbuil Blackbox.
I downloaded Arduino IDE and the "Grbl-master" file, and uploaded it to the arduino library, I can't go on, I don't understand how to edit it!

MARIOBASZ, 2022-03-11:
Were you able to write the file to your arduino?

GrahamJB, 2022-03-11:
I have always found that if I don't understand what I am doing then changing things gives results that are unlikely to be good.

There are many variables to be set in the software configuration.

Nipote, 2022-03-11:
You're right GrahamJB, I'm not up to it, and I think I'll let it go but I also think it's just a crap of a control unit if I can't modify such a parameter myself!!
Which means throwing away a job since if I press the emergency button or open the door afterwards I cannot restart the process!!!
It doesn't seem like a good product to me... what do you think?

GrahamJB, 2022-03-11:
It has to be this way, components change rapidly over time and to adapt and still be of use the software has to have multiple choices. If you can find a configuration for your board then changing a single parameter is much easier.

Nipote, 2022-03-11:
Is there anyone who can make me this change and send me the firmware updated to 10 Seconds ?? So that I just have to load it on the blackbox ??
I can pay for work with PayPal!
I don't know how to do otherwise..

MARIOBASZ, 2022-03-11:
if you were able to record grbl, you can record grbl with that parameter mimidified. My email is mariob_1960@yahoo.com.ar. Send the file and I'll send it back to you corrected. It is free of charge, of course.