gnea/grbl #1130: why only 1 step jog motion is not posible ?

MAK-1966, 2022-04-13:
hello dear forum,

I am trying to move the step motor 1 step with the $J= X0.001----  jog command
however grbl is accumulating the jog commands 
and moves 0.010 when the total of 1 step jog commands is 10 

I want to move 1 step exact (0.001) with jog command 

is this not posible ?

thank you

PicEngraver, 2022-04-15:
I am not sure this is correct, but what do you have your micro-stepping adjustment set to on your stepper driver?  Higher setting values improve stepper resolution response.

MAK-1966, 2022-04-16:
thank you 
if I increase microstep value the speed of the axis decrease - I need fast movement
my question still ; why cannot I move only 1 step ? there must be a setting about it but I couldnot find it

neilferreri, 2022-04-16:
What is your $100 setting?

MAK-1966, 2022-04-16:
hi,
step driver steps/rev is 1000 and $100 attached
[image]