gnea/grbl #1133: Breach of License - Wainlux

CriticalPaint, 2022-04-18:
Sorry I didn't know exactly where to put this so I'm putting it here, I am making whoever is currently the creator / owner of GRBL of a breach to the GPL agreement by the company Wainlux (www.wainlux.com) in relation to thier most recent kickstarter for the 'L3' laser etcher which had some other shady dealings made by the owners during the campaign (saying they would supply a cracked version of the Lightburn software....) have supplied along with thier own firmware, a compiled GRBL firmware for use with lightburn. I have made multiple attempts via multiple means to get the configuration files that were used to compile the GRBL firmware and have met only resistance in the process, even after mentioning that they are in breach of the GPL agreement. the latest response from them being "Hello, we have our own software, open source software is not our software, we cannot provide data and parameters, our machines can only use this software." which is straight up lies, they supplied a usb drive with the GRBL frimware and it is also available to download directly from thier site. Usually I would give someone the benefit of the doubt but these guys have been asking for it... *ahem* "Let slip the dogs of war!" if you need to contact me send me an email,

CriticalPaint, 2022-04-25:
wow, nothing? nothing at all?

arkypita, 2022-04-26:
Hi, I am the developer of LaserGRBL (not GRBL)
The behavior you describe is unfortunately the norm with regard to engraver manufacturers. Almost all of them use versions of GRBL firmware, modify them, and do not release the source (sometimes not even precompiled binaries).

I think the developers of GRBL are fully aware of the phenomenon, but I doubt they have any interest/willingness to fight it, as it is kind of a losing battle. It interests much more to those of LightBurn that being commercial software it has an economic damage if someone distributes cracked copies.

In any case, I would like to know more as I have received requests for collaboration from Wainlux, and I would like to avoid having to deal with them if they have these behaviors with customers. If possible, please forward to me your emails with them

My email: info@lasergrbl.com

HomoFaciens, 2022-07-01:
I am running a YouTube channel dedicated to technology and I got laser engravers offered for reviews. I have started with an Ortur LaserMaster 2 2S and after recording the video I requested the modified source code, since it is obvious that their firmware is based on grbl. First answer: nothing. I then had a second attempt though the sales department that sent me the laser machine. Answer: Their firmware is "business secret".
I am now willing to play that game a bit longer, since influencers have a louder say than simple customers. When FDM printers from China flooded the market, they also ignored the license of Marlin. A bit of pressure from influencers on YouTube have changed that in large parts. Some manufacturers still ignore it, but I'd say the majority somehow follows the rules.

**So my request to the grbl developers is: Is that kind of fight for your rights okay for you? A quick YES / NO answer would make me feel better / stop me wasting my time.**

Thanks!

chamnit, 2022-07-01:
Yes. I’m fine with it. Although in general, you don’t have to seek permission. Grbl is in the public domain and licensed as such.

AlexHolden, 2022-07-02:
> Yes. I’m fine with it. Although in general, you don’t have to seek permission. Grbl is in the public domain and licensed as such.

As I understand it, Grbl is protected by the GPL v3.0 license.

Public domain is different; it means something is not protected by copyright at all, either because the author explicitly stated that they don't wish to copyright the work or because the copyright term has expired.

HomoFaciens, 2022-07-02:
> As I understand it, Grbl is protected by the GPL v3.0 license.
>
> Public domain is different; it means something is not protected by copyright at all, either because the author explicitly stated that [...]

Yes, according to the file "COPYING" on the grbl repository it is GPL v3 or later and not public domain. This is why I would like to insist on publishing the modified sources of the laser machine manufacturers and enables me to do so in terms of law.

@chamnit thanks for voting.

CriticalPaint, 2022-07-02:
I have attempted to get the source from them multiple times and they keep coming back with the "business secret" crap, I have since been investing my time into rebuilding the machine around a makerbase board and have been having far more success in obtaining what it wanted / needed from the original product, but I will still keep up with this @HomoFaciens and would like to know how far you can get with them!

HomoFaciens, 2022-07-09:
I now have a couple of laser cutters from various manufacturers that I got for free to make reviews. I will do so, but of course have a look if they make use of grbl. If so, before publishing a video I ask for the source code.

My personal escalation levels are:
(1) Ask the support or sales team for the source code. If answer is "no":
(2) Ask again with clear hints to the license violation. If answer is "no":
(3) Publish a video on my channel that is not the review they want to see.

Ortur is the first manufacturer on my list and unfortunately also the first one that reached escalation level 3. The video is now live on my YouTube channel:
[https://youtu.be/ekITN_m5mKs](url)

It is my new channel, dedicated to the openness of gadgets, with not so many subscribers, yet, but it will also go live on my bigger channel.

Let's see it that can change the mind of Ortur and other black sheep.
Any remarks on the video are welcome!

HomoFaciens, 2022-07-16:
Tides are turning!
Ortur sent me a roadmap of how and when they will open the grbl related Source Code in the coming weeks. 
If anyone of the grbl team is interested to get in contact with the man of the software department that now is responsible to make that happen, I can point to this page.
Yes? / No?

iCodeSometime, 2022-07-17:
Frankly, a roadmap to release it in the future is unacceptable. They are only allowed to distribute the software at all through compliance with the license terms.

They should either immediately release the source that should have been available since the beginning, or else immediately cease all sales. Refusing to comply with the license terms is theft.

Maybe the Software Freedom Conservancy may be willing to get involved if any of the copyright holders are willing to work with them.

HomoFaciens, 2022-07-17:
> Frankly, a roadmap to release it in the future is unacceptable. They are only allowed to distribute the software at all through compliance with the license terms.
>
> They should either immediately [...]

Cool down. The roadmap isn't somewhen in the far, far future, but talks of a relatively short time span. Give Ortur some time to make things right instead of just copy & paste some code lines somewhere on a ghost town in the Internet.

The man I am in contact with simply wants to know how to make things in a way that it will actually be useful for the community. He was searching for other common manufacturers of laser machines to see how they are handling the stuff and _surprise_ he found none! So don't throw your (understandable) anger in the face of the (only?) one that works on a change.

iCodeSometime, 2022-10-13:
It is good that they intend to fix the problem, but not good that their plan is to continue to steal labor from the community until they work out the details on their own timeline. I see source for some older models has been [uploaded now](https://github.com/Ortur-Tech/Ortur_Laser_Firmware_17X-18X) - that's great!

"Copy/Paste some code lines on a ghost town in the internet" is a perfectly acceptable solution as long as anyone using their software has access to it. Providing it as a zip when requested by email, or even mailing out USB drives at cost are also perfectly valid ways to comply with the license (see section 6) - it is very easy to comply.

The license is very clear that they have no rights to distribute binaries without also providing the source. There's no good excuse to continue to distribute software they have no right to (especially when the bar for obtaining that right is so low).

Of course you are right, it's worse that others have no intention of ever complying with the license requirements. I really appreciate your efforts in this project, and I'm glad you are making progress. I'll check out some more of your videos 👍

HomoFaciens, 2023-04-13:
With FlyingBear, another manufacturer now follows the rules and published the modified source code:
[](https://drive.google.com/drive/folders/1SOQ87Hw8J7gjQCtrIzGSMM4Fh3et-Jet)