gnea/grbl #1134: #define SOME_DEFAULT does not change anything

BernardG, 2022-04-18:
Don't understand this one. It looks like, whatever I do, I Find ALWAYS the generic defaults after I compile. I have created my own, and tried with some existing defaults, like Sherline 5400, but nothing changes.  Does this have to do with overwriting EEPROM? And, if yes, how should I do that?

neilferreri, 2022-04-18:
#Does this have to do with overwriting EEPROM?
Yes. You can either just write the settings individually by sending the new $$ values or you can clear the EEPROM using the built in Arduino program and upload Grbl with your defaults.

BernardG, 2022-04-19:
> > Does this have to do with overwriting EEPROM?
> 
> Yes. You can either just write the settings individually by sending the new $$ values or you can clear the EEPROM using the built in Arduino program and upload Grbl with your defaults.

OK, thank you. I will use the Arduino built-in program to be sure.