gnea/grbl #1141: A few questions from somebody new to CNC machines

VaasKahnGrim, 2022-05-22:
So I have a goal in mind for building my custom CNC machine. I want to be able to have 5-axis of motion(not because I NEED it right now but because I MAY need it later), I'm also wanting to have it do more than just taking a block of what ever material and then cutting into it. I want to be able to switch between multiple tools on the same table automatically without manual intervention required, or atleast not BY HAND. I want to be able to switch between a drill bit, a wire brush, sand paper, or maybe even having a tool head that could be mounted to function as a 3D printer, so it would add material instead of taking away material.

SO thats my goal, and I know that technically everything I'm about to ask can be done. my questions are more regarding if these questions I'm asking are already implemented or if I'd have to implement them myself.

So lets actually ask the questions:
1. Does GRBL have what I'd need to build a 5 axis CNC already in it?
2. Is there a way to tell my CNC when to switch tools and to which tool it should switch to? (hopefully being able to automate when that should happen aswell)
3. Does GRBL support a sort of plugin system or API for communicating witht he computer and the actual CNC table?(like being able to see what step in the process it's at, elapsed time, temprature, motor speeds, ect)
4. Can it count how many jobs of a specific kind it has done?(so like if you want it to do exactly 50 itterations of the same job, it would be able to do it and get a new block of material for each itteration of the job before moving onto the next job)
5. Can GRBL interact with other parts besides the servos and tool peices? (could it opperate a convayor belt, could it opperate the tool peice changing mechanism, could it opperate an arm peice that moves the object after its been finished working on, ect)

I know these questions might be for alot more than your typical at home use CNC machine and most might not need that sort of thing but I would like to automate alot of the actual work going into alot of projects I have so being able to do these things would be very useful for me personally. I just want to know how much of this is already feasable with GRBL and how much I'd actually have to program myself

MARIOBASZ, 2022-05-23:
Maybe I'm wrong, this is what I think:
1- There is a version of grbl for 5 axes, it runs on Arduino Mega. It's what I'm using now. There is also grblHal (on 32-bit boards, like Due. I'll migrate there)
2- grbl does not support tool change. However, using bCNC, you can use command M6 T x (x = tool number) for manual tool change.
So far what I know.
Hugs.

terjeio, 2022-05-23:
For [grblHAL](https://github.com/grblHAL) the answers will be:

1. Yes.
2. Yes, but AFAIK only [ioSender](https://github.com/terjeio/ioSender) supports the grblHAL [tool change protocol](https://github.com/grblHAL/core/wiki/Manual,-semi-automatic-and-automatic-tool-change). Plugin code can add support for automated tool changes.
3. [Yes](https://github.com/grblHAL/plugins).
4. Yes, if you write the needed plugin code.
5. Same as 4.

langwadt, 2022-05-23:
no it will not run on a PC, in short a PC doesn't have any suitable IOs

VaasKahnGrim, 2022-05-23:
what do you mean by IOs? are you refering to like interfaces(USB/PCIe/SATA) or like actual programable jumper pins? Or a designated interface like serial port or something along those lines?

Would you have a recomendation of what I'd need to support running grblHAL since it would be x86 architecture?

terjeio, 2022-05-24:
You can find a list of most of the supported controllers [here](https://github.com/grblHAL/Controllers) and processor/driver features [here](https://github.com/grblHAL/drivers).

BTW the code will run on x86 but only in [simulation mode](https://github.com/grblHAL/Simulator) - this driver cannot control a CNC machine.

langwadt, 2022-05-24:
@VaasKahnGrim actual pins that can be toggled in consistent realtime, not since ancient times when PCs had parallelports have  PCs had that. and even if they had it would be able to do it in realtime without a special OS and or major hacks to the OS

terjeio, 2022-05-24:
> and even if they had it would be able to do it in realtime without a special OS and or major hacks to the OS

Mach3 manages to do that with Windows (but hard to get working on a laptop), and LinuxCNC with a real-time version of Linux. There are of course dedicated controllers for these that bypass the parallell port - but they are more expensive than a grbl controller...

langwadt, 2022-05-25:
yes, LinuxCNC does it with a special OS and Mach3 does it by installing a custom ring0 kernel driver on some HW timer to run the step generator