gnea/grbl #1144: Is there a bottleneck in processing the planning buffer?

svenhb, 2022-05-30:
Is there a bottleneck in processing the planning buffer?
If line-length is too short, plannig buffer will not be filled completly -> increasing processing time...
In my tests the limit seems to be at 0.3 mm segment length - example 2).
If I try with 0.2 mm segment length but compressed code - example 4) (less bytes than with 0.3 mm segment length) the processing time is longer.
So the communication speed can't be the issue (as thought in #1036).

I have a test graphics (Test_Arc_1000mm.dxf) - a meandering arrangement of semicircles. 
[Test_Arc.zip](https://github.com/gnea/grbl/files/8799108/Test_Arc.zip)

All following tests uses same grbl-settings and same XY feedrate of 5000 mm/min, no Z axis or other functions are involved...
Arduino-Nano (stand alone) with grbl 1.1f without connected hardware.

Imported with my GRBL-Plotter program:
1) Exported to G2 / G3 commands (= 5 code lines), the exceution time is 19 seconds.
2) Exported to line segments with a minimum length of 0.3 mm (= 3929 code lines; 94604 chars), the exceution time is still 19 seconds.(Test_Arc_03_normal.nc)
during run, available blocks in planner buffer stays at 0
3) Exported to line segments with a minimum length of 0.2 mm (= 6449 code lines; 156813 chars), the exceution time is 24 seconds.(Test_Arc_02_normal.nc)
during run, available blocks in planner buffer goes up to 12
4) Exported to line segments with a minimum length of 0.2 mm but 2 decimals, relative movements, compressed (= 6449 code lines; 77722 chars), the exceution time is 22 seconds.(Test_Arc_02_optimized.nc)
during run, available blocks in planner buffer goes up to 10

Can anyone confirm this behavior?

Screenshot, when streaming Test_Arc_02_optimized.nc, the planning buffer shows 8, and 44 bytes left in the RX buffer:
[image]

GreenLightning, 2022-06-07:
This looks like a throughput issue, but not necessarily related to the processing of the planning buffer.

The RX buffer is 128 bytes, so if there are 44 bytes available, there are 84 bytes stored. At 12 bytes/line (including new line) that is exactly 7 lines. So, it is not the communication speed, but the throughput from RX buffer to planner buffer.

The RX buffer is read [here](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/protocol.c#L78) in `protocol_main_loop`, which does a little bit of preprocessing and then calls `gc_execute_line` [here](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/protocol.c#L104). [`gc_execute_line`](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/gcode.c#L66) has to parse the line into a G-Code block first, which is quite a bit of work (basically all of `gcode.c`) and then eventually calls `mc_line` [here](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/gcode.c#L1053). Then `mc_line` calls `plan_buffer_line` [here](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/motion_control.c#L68). [`plan_buffer_line`](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/planner.c#L315) has to do a bunch of calculations to fill the new planner block, which is added to the planner buffer and then also calls `planner_recalculate` [here](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/planner.c#L469). [`planner_recalculate`](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/planner.c#L126) updates all the blocks in the planner buffer.

In summary, all of this code has to run for every line of g-code / every line segment. If you use shorter line segments, there are more line segments, so more work to do. At some point you are limited by the speed of the processor.

P.S.: Excellent test setup to rule out the communication speed.

GreenLightning, 2022-06-07:
There's a different issue with the planner buffer, in that `planner_recalculate` limits the speeds so that the machine is able to stop at the end of the buffer. This depends on the distance stored in the planner buffer (which is the number of blocks times the segment length (assuming all segments have the same length)) and the acceleration settings.

I have a G-Code simulator and using a lower acceleration of 200 mm/s^2 the predicted execution times are 20s for G2G3, 33s for 03_normal and 41s for 02_optimized. However, using the GRBL settings you provided (2000 mm/s^2 acceleration) predicts 20s for all three files, so this is not an issue (assuming the planner buffer can always be filled).

svenhb, 2022-06-09:
Thanks for the detailed analysis.