gnea/grbl #1147: StepperOnline DM860T stepper drivers hissing but moving after half a second

bbuster2, 2022-06-17:
I'm sorry for my bad title but I couldn't think of a better one.

I am using StepperOnline DM860T drivers connected to a GRBL controller.
Previously I used cheap TB6600 drivers which worked instantly, but after upgrading I am having problems manually moving the motors, and by manually I mean with the controller but not in a program.

When I press the button to move the X-axis for example, the motors hiss for half a second and than move (at least if I move them enough mm), if I move them for example 10mm they only hiss without moving. When I move them for 100mm they his for half a second and then move for a remaining 70mm for example.

I just figured out that if I set the $1 command to 255 (Sets the time delay in milliseconds that GRBL will power the stepper motors after a motion command is complete. A setting of 255 tells the motors to stay powered on to hold position.) the motors work perfectly fine and respond immediately. The only problem is that the controller always sees that the motors are enabled and I cannot do anything because I need to wait till the motors are finished.

Also the motors are enabled all the time with this setting which is probably not a good thing.
The StepperOnline DM860T have a few settings (see attachement).

The only problem is I cannot find a GRBL setting to make the motors enable more quickly or have a delay after the enabling before trying to move.

How can I solve this?

PS: The controller I am using is a Makerbase MKS DLC32
[image]

MARIOBASZ, 2022-06-18:
Just a guess: You will not have very high acceleration, or very low current, and the motor does not respond as it should, losing steps, and hence the hiss?

bbuster2, 2022-06-18:
I played with acceleration speed but it didn’t change anything. Now it is set to 500 i believe. 

The motors are 2.8A and the drivers can go up to over 8A i believe. I also tested with higher current settings but the result was the same