gnea/grbl #1149: Wiring optical endstop

ScottFal, 2022-07-02:
when i wire up the optical endstop like the picture, it works the opposite as I would like, it gets detected with nothing blocking it, and isn't detected with something in the way, how would I go around changing it
[image]

optical endstop used: https://www.amazon.co.uk/gp/product/B077X8XL56/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

langwadt, 2022-07-02:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#5----limit-pins-invert-boolean

ScottFal, 2022-07-03:
> https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#5----limit-pins-invert-boolean

this does work but, it gives me a new problem, the pins without limit stops(z) now always alarm, is there a way I can ignore the x axis alarm?

closed 2022-07-03 by ScottFal