gnea/grbl #1157: Would like to clarify how the door security function operates.

GForce2010, 2022-08-05:
I have a small PCP CNC running GRBL 1.1. For safety reasons it sits in an enclosure and I am considering adding a door switch to ensure the machine cannot be run with the door open.

My concern is that I have believe if I have a door switch enabled the machine cannot be jogged with the door open. Is that the case?

In a perfect world, I would want the ability to open the door, jog the machine into position and probe the surface etc, but not be able to start the milling process until the door is closed.

Is this something the is possible to do in GRBL?

Thanks.

langwadt, 2022-08-05:
everything is possible, the code is right here for you to modify as you want

mar0x, 2022-08-05:
Hello.
Currently the jogging is also blocked when safety door enabled and opened.
The modification suggested by @langwadt is possible, but doesn't looks trivial.

terjeio, 2022-08-11:
I've added a settings flag for [ignoring the door signal when in IDLE mode](https://github.com/grblHAL/core/blob/53855a4bf8a2c31fae417c692d512cd10af00b80/system.c#L110-L119) in my 32-bit port, likely alongside a check blocking anything beside jogging when the door is open.