gnea/grbl #1158: GRBL Disconnecting randomly during a job

hardingb4, 2022-08-11:
I am having the issue of GRBL disconnecting randomly during a job. I make crib boards so I use the same gcode over and over, attached is the gcode for drilling the 367 holes needed for one board. It randomly will stop in the middle because the machine disconnects. I have to disconnect the USB and plug it back in before I can reconnect to the machine. By random, I mean it happens after 10 holes sometimes, sometimes it finished two boards. Almost always it disconnects while moving from hole to hole at rapid speed.

My setup:
Homemade cnc with Arduino Uno with shield running A4988 drivers. 500w air cooled spindle.
Power for the Arduino is from a 24V power supply and power for the spindle is from its own power supply. Both power supplies use the same power cord to get power.

What I have tried:
- Replaced USB cable, new one is shielded.
- Reduced max rate on all axis's way down to 1000mm/min
- Tried a different computer to run the gcode
- separated spindle power wires from stepper wires
- Replaced Arduino with a new Arduino
- Tried UGS and Gsender
- Tried powered usb hub

I currently have ferrites on order, where should I put these. I believe I should put 1 on each end of the USB cable, should I put them on each stepper cable and spindle power cable?

I have multiple 3D printers and laser engravers and have never had this issue with those. Is there anything else anyone can recommend to try? I have thought about using shielded cat 6 cables for the stepper motors, would this help? How about changing the power so that I plug in different cables into different circuits for the 24V power supply and spindle power supply? Should I be grounding the Arduino in some way?

Below is my grbl config and attached is the file I commonly have issues with.

$0 | 10 | Step pulse time
$1 | 255 | Step idle delay
$2 | 0 | Step pulse invert
$3 | 0 | Step direction invert
$4 | 0 | Invert step enable pin
$5 | 0 | Invert limit pins
$6 | 0 | Invert probe pin
$10 | 1 | Status report options
$11 | 0.010 | Junction deviation
$12 | 0.002 | Arc tolerance
$13 | 0 | Report in inches
$20 | 0 | Soft limits enable
$21 | 0 | Hard limits enable
$22 | 0 | Homing cycle enable
$23 | 0 | Homing direction invert
$24 | 25.000 | Homing locate feed rate
$25 | 500.000 | Homing search seek rate
$26 | 250 | Homing switch debounce delay
$27 | 1.000 | Homing switch pull-off distance
$30 | 1000 | Maximum spindle speed
$31 | 0 | Minimum spindle speed
$32 | 0 | Laser-mode enable
$100 | 40.000 | X-axis travel resolution
$101 | 40.000 | Y-axis travel resolution
$102 | 250.000 | Z-axis travel resolution
$110 | 1000.000 | X-axis maximum rate
$111 | 1000.000 | Y-axis maximum rate
$112 | 500.000 | Z-axis maximum rate
$120 | 1000.000 | X-axis acceleration
$121 | 1000.000 | Y-axis acceleration
$122 | 1000.000 | Z-axis acceleration
$130 | 200.000 | X-axis maximum travel
$131 | 200.000 | Y-axis maximum travel
$132 | 200.000 | Z-axis maximum travel

[Drill.zip](https://github.com/gnea/grbl/files/9308871/Drill.zip)

bobbybumfluff, 2022-08-11:
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "https://github.com/gnea/grbl/issues/1158", "name": "View Issue" },
> "Organization", "name": "GitHub", "url": "https://github.com" } } ]

Looks like electrical interference from motor wiring.

daniJVV, 2022-08-26:
Agreed with bobbybumfluff, did you tried a USB isolator between your machine and the PC?

closed 2022-08-31 by hardingb4