gnea/grbl #1159: Homing fail

LeeD123456, 2022-08-23:
hello guys, I am building an engraving machine with 2 axes. I have a problem with setup home. 
I have followed step-by-step tutorials on youtube but failed (detail below image). 
Could you give me how to fix this problem?
[image]
[image]
 thank all

lalo-uy, 2022-08-23:
The switch near the head must be mounted over the vertical axe. Then it
will move with it.

lalo-uy, 2022-10-11:
> Thanks for your idea so much. But I really don't get it. Could you explain
> detailly or describe by image

Mount the Home switch over the axis.